jeremiah-c-leary/vhdl-style-guide #1130: procedure_call_003 alignment glitch with positional parameters

alonbl, 2024-02-21:
**Environment**

```
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.4 LTS"
$ vsg --version
VHDL Style Guide (VSG) version: 3.21.0+zip.file
Git commit SHA: Unknown.  Installed via zip file.
```

**Describe the bug**

Looking at [1]
I would like to reach to the example of `test_harness` when using positional parameters. However, when using positional parameters the alignment misbehaved, I could not find any combination of settings to allow both positional and named parameters to work in a similar structure. Maybe it is on purpose not sure...

```
        test_hardness(
            enabled  => true,
            sequence => "test_001",
            logging  => "full"
        );
        test_hardness(true, "test_001", "full"
        );
```

[1] https://vhdl-style-guide.readthedocs.io/en/latest/configuring_procedure_call_statement_rules.html#configuring-procedure-call-statement-rules

**To Reproduce**

*VHDL*

Baseline is the following:

```
architecture behavioral of b1 is
begin
    test_hardness(enabled=>true, sequence=>"test_001", logging => "full");
    test_hardness(true, "test_001", "full");
end architecture behavioral;
```

*Configuration*

```yaml
rule:
  global:
    align_left: 'yes'
    align_paren: 'no'
    align_when_keywords: 'yes'
    association_element: add_new_line
    association_list_comma: remove_new_line
    first_open_paren: remove_new_line
    first_paren_new_line : 'no'
    last_close_paren: add_new_line
```

**Expected behavior**

```
        test_hardness(
            enabled  => true,
            sequence => "test_001",
            logging  => "full"
        );
        test_hardness(
            true,
            "test_001",
            "full"
        );
```

jeremiah-c-leary, 2024-02-29:
Morning @alonbl ,

I pushed an update for this to the `issue-1130` branch.  When you get a chance could you check it out on your end and let me know if it resolves the issue.

Thanks,

--Jeremy

alonbl, 2024-03-02:
Hi @jeremiah-c-leary,
Working perfectly.
Thank you so much!
Alon

jeremiah-c-leary, 2024-03-03:
Awesome, I will get this merged to master.

--Jeremy

PR #1134 (2024-03-03): Issue#1130:  Including parameters without a formal part in rule proce…
Resolves #1130
closed 2024-03-03 by jeremiah-c-leary