jeremiah-c-leary/vhdl-style-guide #1135: Add documentation: Rule is (auto)fixable

el-coder-sb, 2024-03-07:
**Is your feature request related to a problem? Please describe.**
I can´t find any information about if there are any "Configuring Prefix and Suffix Rules" which are fixabel. Or better said: Would be helpful to have the information if a rule is (auto)fixable and if yes under which circumstances (regarding suffixes: surely only if only one allowed suffix ist given)

**Describe the solution you'd like**
A flag or text in the documentation which indicates if a rule is (auto)fixable.

**Additional context**
Somewhat related to #558.

jeremiah-c-leary, 2024-03-07:
Morning @el-coder-sb ,

I see your point.  I believe there are several rules where the documentation states the user is responsible for fixing the issue.  However, being explicit with every rule would be helpful.

I would propose adding another icon...something like:  

[image]

to each rule that VSG can not fix.  This would then link to some new documentation describing the option.

It would be similar to rules that are disabled.  Most rules are enabled default so there is not icon.  The disabled rules, e.g. `package_body_600` have an icon indicating it is disabled.

Would that approach be satisfactory?

Regards,

--Jeremy

jeremiah-c-leary, 2024-03-09:
Morning @el-coder-sb ,

I pushed the documentation update to the `issue-1135` branch.  You can build the documentation by following these steps:

```mono
$ cd docs
$ make html
```
The results will be stored in the `_build` directory.
Load the `index.html` file in your browser.

Let me know if you think there are any changes I should make.

Regards,

--Jeremy

el-coder-sb, 2024-03-21:
Thank you very much for your quick solution. Sorry for late response. I have some trouble with my VM where I use VSG. At the moment I´m not able to test. I´ll come back when I got time to fix my VM and test VSG.

jeremiah-c-leary, 2024-04-02:
Morning @el-coder-sb ,

Just a ping to see if you were able to check out the updated documentation.

--Jeremy

jeremiah-c-leary, 2024-04-17:
Evening @el-coder-sb ,

No problem.  I will get this merged to master.

--Jeremy

PR #1136 (2024-04-17): Issue#1135:  Adding documentation for rules which can not be automati…
Resolves #1135
closed 2024-04-17 by jeremiah-c-leary