jeremiah-c-leary/vhdl-style-guide #1143: docs: Typos

vornkat-iis, 2024-03-18:
Some minor errors in the docs which took me some minutes to figure out:

## docs/tool_integration/pre-commit.rst
Should be (without a v in the tag) two times on that page:
```
repos:
  - repo: https://github.com/jeremiah-c-leary/vhdl-style-guide
    rev: 3.18.0
    hooks:
      - id: vsg
``` 

# docs/tool_integration/generic.rst
Should be `--all_phases` instead of `--all-phases`.

Just some nits to a tool which was great to setup and we are so far very happy to use! Thank you for providing it!

jeremiah-c-leary, 2024-03-18:
Evening @vornkat-iis ,

> Just some nits to a tool which was great to setup and we are so far very happy to use! Thank you for providing it!

Thanks for the kind words.  It's nice to hear that my work is appreciated.

I committed the change to the `issue-1143` branch.  When you get a chance could you check out the documentation updates on your end and let me know if it looks good to you?

Thanks,

--Jeremy

vornkat-iis, 2024-03-19:
Hi @jeremiah-c-leary,

yes, this is exactly what I meant, thanks for the fast response! Approved :)

Happy coding!

jeremiah-c-leary, 2024-03-19:
Morning @vornkat-iis ,

Awesome, I will get his merged to master.

--Jeremy

PR #1144 (2024-03-19): Issue#1143:  Documentation updates
Resolves #1143
closed 2024-03-19 by jeremiah-c-leary