jeremiah-c-leary/vhdl-style-guide #1145: Configuration JSON format is not JSON compliant

alonbl, 2024-03-24:
Hi,

Remaining issue from #1141.

```json
  {
    "file_list": [
     "tests/vsg/read_configuration_files/entity.vhd",
      "tests/vsg/read_configuration_files/arch.vhd": {          <---- json cannot have dict key in a list
        "rule": {
          "rule_001": {
            "disable": true
          }
        }
      },
     "tests/vsg/read_configuration_files/package.vhd"
    ]
  }
```

I suggest the following:

if the value is string it is a name, if it is a dictionary take name from name key:

```json
  {
    "file_list": [
     "tests/vsg/read_configuration_files/entity.vhd",
      {
        "name": "tests/vsg/read_configuration_files/arch.vhd",
        "rule": {
          "rule_001": {
            "disable": true
          }
        }
      },
     "tests/vsg/read_configuration_files/package.vhd"
    ]
  }
```

jeremiah-c-leary, 2024-04-10:
Evening @alonbl ,

I reviewed the JSON standard and found out that a list can contain a string, number, object, array, true, false, or null.  I am trying to define an object but I am missing the opening and closing curly braces.  So the above file can be updated to this:

```json
  {
    "file_list": [
      "tests/vsg/read_configuration_files/entity.vhd",
      {
        "tests/vsg/read_configuration_files/arch.vhd": {
          "rule": {
            "rule_001": {
              "disable": true
            }
          }
        }
      },
      "tests/vsg/read_configuration_files/package.vhd"
    ]
  }
```

I updated the two JSON files that were failing pre-commit and updated the documentation to include the curly braces.

I pushed an update to the `issue-1145` branch.  When you get a chance could you check it  on your end.

Thanks,

--Jeremy

alonbl, 2024-04-10:
Hi @jeremiah-c-leary ,

It is nice that this notation is supported without a code change :)
And that we are not compliant with json.

However, I think the format is confusing.

Current notation, assumes generic key name (aka file) within a single dictionary.

```json
{
    "file_list": [
        {
            "file1": {
                "key": "value"
            }
        },
        {
            "file2": {
                "key": "value"
            }
        },
        "file3"
    ]
}
```

Alternative:

```json
{
    "file_list": {
        "file1": {
            "key": "value"
        },
        "file2": {
            "key": "value"
        },
        "file3"
    }
}
```

The alternative is simpler, however, the order of a map is not fixed, I believe that in this case the order matters.

In order to have proper schema, I suggested the following, as each item in the list is well defined:

```json
{
    "file_list": [
        {
            "name": "file1",
            "rule": {
                "key": "value"
            }
        },
        {
            "name": "file2",
            "rule": {
                "key": "value"
            }
        },
        "file3"
    ]
}
```

Also in yaml, as the key names are not wildcards no need to play with yaml magic:

```yaml
file_list:
   - name: file1/**
     rule:
       key: value
   - name: file2
     rule:
       key: value
   - file3
```

The interesting variation of the above is the ability to provide multiple names, which is impossible in the current one.

```json
{
    "file_list": [
        {
            "name": [
                "file1/**",
                "file1.1/**"
            ],
            "rule": {
                "key": "value"
            }
        },
        {
            "name": "file2",
            "rule": {
                "key": "value"
            }
        },
        "file3"
    ]
}
```

```yaml
file_list:
   - name:
     - file1/**
     - file1.1/**
     rule:
       key: value
   - name: file2
     rule:
       key: value
   - file3
```

Branch `issue-1145` is great in term of making json pass, just wanted to write why the current notation is a bit complex and limiting.

Regards,

jeremiah-c-leary, 2024-04-12:
Morning @alonbl ,

Would you be okay with merging this issue to resolve the error and create another one to discuss changes to configurations?  There may be other improvements that could be made.

--Jeremy

alonbl, 2024-04-12:
Sure. It solves the incompatibility.
Thank you.

jeremiah-c-leary, 2024-04-13:
Sweet, I will merge this to master.

--Jeremy

PR #1165 (2024-04-13): Issue#1145:  Fixing invalid JSON test files and documentation.
Resolves #1145
closed 2024-04-13 by jeremiah-c-leary