jeremiah-c-leary/vhdl-style-guide #1146: Remove invalid escape sequences from docstring

alonbl, 2024-03-24:
Remaining issue from #1141.

Resolve the following warning during build, probably escape during `rst` generation instead.

```
DeprecationWarning: invalid escape sequence XXXX
```

jeremiah-c-leary, 2024-03-25:
Evening @alonbl ,

I pushed an update for this to the `issue-1146` branch.  When you get a chance could you check on your end that the issue is resolved.

For some reason I get the following reported by pytest:
```mono
/home/jcleary/projects/vsg-master/.tox/test-py38/lib/python3.8/site-packages/_pytest/terminal.py:876: PluggyTeardownRaisedWarning: A plugin raised an exception during an old-style hookwrapper teardown.
Plugin: 140489805125376, Hook: pytest_terminal_summary
JSONDecodeError: Expecting value: line 1 column 49128 (char 49127)
For more information see https://pluggy.readthedocs.io/en/stable/api_reference.html#pluggy.PluggyTeardownRaisedWarning
  self.config.hook.pytest_terminal_summary(
......../home/jcleary/projects/vsg-master/.tox/test-py38/lib/python3.8/site-packages/_pytest/terminal.py:876: PluggyTeardownRaisedWarning: A plugin raised an exception during an old-style hookwrapper teardown.
Plugin: 140674028668672, Hook: pytest_terminal_summary
JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 16371 (char 16370)
For more information see https://pluggy.readthedocs.io/en/stable/api_reference.html#pluggy.PluggyTeardownRaisedWarning
  self.config.hook.pytest_terminal_summary(
```

Everything seems to run, so it is confusing why it shows up.

--Jeremy

alonbl, 2024-03-25:
Hi,

The ci[1] passed also I checked it and it works at my end, so it must be something locally.
Can you please verify that you have no important files within your git workspace and:

```sh
git reset --hard
git clean -dxff
tox
tox -e docs
```

[1] https://github.com/jeremiah-c-leary/vhdl-style-guide/actions/runs/8413651869

jeremiah-c-leary, 2024-03-26:
Evening @alonbl ,

I ran the commands you listed above and check for any uncommitted files.  Interestingly the error does not show up when I run `tox -e test-py8`, but it does show up when I run `tox -e py310`.

--Jeremy

alonbl, 2024-03-26:
`tox -e py310` should not do anything as there is no environment at `py310` name, see below output.

```
$ tox -e py310
ROOT: tox-gh-actions won't override envlist because tox is not running in GitHub Actions
.pkg: _optional_hooks> python /home/alonbl/tmp/vhdl-style-guide/.tox/.tox/lib/python3.10/site-packages/pyproject_api/_backend.py True setuptools.build_meta
.pkg: get_requires_for_build_sdist> python /home/alonbl/tmp/vhdl-style-guide/.tox/.tox/lib/python3.10/site-packages/pyproject_api/_backend.py True setuptools.build_meta
.pkg: build_sdist> python /home/alonbl/tmp/vhdl-style-guide/.tox/.tox/lib/python3.10/site-packages/pyproject_api/_backend.py True setuptools.build_meta
py310: install_package> python -I -m pip install --force-reinstall --no-deps /home/alonbl/tmp/vhdl-style-guide/.tox/.tmp/package/2/vsg-3.22.0.post27+git.b4e1d2d0.tar.gz
  py310: OK (12.55 seconds)
  congratulations :) (12.69 seconds)
```

I see the error is of python-3.8.

Is this reproduced always?
Can you please try:

```diff
diff --git a/tox.ini b/tox.ini
index f287ac97..da89313b 100644
--- a/tox.ini
+++ b/tox.ini
@@ -34,6 +34,7 @@ setenv =
     COVERAGE_DATA_FILE=build.out/.coverage.{basepython}
 commands =
     pytest \
+        --tb=long -v \
         --cov-report=html:build.out/{envname}/coverage \
         --cov-report=xml:build.out/{envname}/coverage.xml \
         --html-report=build.out/{envname}/report/pytest-report.html \
```
```

jeremiah-c-leary, 2024-03-27:
Evening @alonbl ,

With `--tb-long -v` added, the error message does not appear.

That is strange though.

--Jeremy

alonbl, 2024-03-27:
Hi @jeremiah-c-leary,
Just to make sure, without the verbose you can reproduce this 100% with python-3.10, correct?
And you can reproduce this only after the changes in this branch and not in master.
I reviewed the changes many times, I cannot understand what triggers this in this branch so it must be reproducible also in master.
It suggests that I missed some resource free/reuse between tests that cause conflict and exception, probably at setup or teardown.
Alon

PR #1152 (2024-03-28): Issue#1146:  Fixing invalid escaped characters in documentation and tests
Resolves #1146
closed 2024-03-28 by jeremiah-c-leary