jeremiah-c-leary/vhdl-style-guide #1163: `Configuring Use Clause Indenting` documentation issue

avelure, 2024-04-04:
**Environment**
v3.22.0 win11

**Describe the bug**
In https://vhdl-style-guide.readthedocs.io/en/stable/configuring_use_clause_indenting.html#configuring-use-clause-indenting it is specified we should use 
```
indent:
  tokens:
     use_clause:
       token_after_library_clause: '+1'
       token_if_no_matching_library_clause: '+1'
```
but this causes a traceback
```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\asd\AppData\Local\Anaconda3\Scripts\vsg.exe\__main__.py", line 7, in <module>
  File "C:\Users\asd\AppData\Local\anaconda3\Lib\site-packages\vsg\__main__.py", line 110, in main
    oConfig = config.New(commandLineArguments)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\asd\AppData\Local\anaconda3\Lib\site-packages\vsg\config.py", line 316, in New
    dIndent = read_indent_configuration(dConfig)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\asd\AppData\Local\anaconda3\Lib\site-packages\vsg\config.py", line 211, in read_indent_configuration
    for sParameter in list(dGroups[sGroup][sToken].keys()):
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'int' object has no attribute 'keys'
```
according to #1030 it is missing `keyword` and should be
```
indent:
  tokens:
    use_clause:
      keyword:
        token_after_library_clause: '+1'
        token_if_no_matching_library_clause: '+1'
```

jeremiah-c-leary, 2024-04-04:
Morning @avelure ,

Thank you for finding the documentation error.  I have pushed an update to the `issue-1163` branch.  When you get a chance could you check on your end that the update is correct.

I will plan to merge this to master before the next release, which should hopefully be this weekend.

Thanks,

--Jeremy

avelure, 2024-04-04:
I had a look at the diff and it looks ok, thanks :)

avelure, 2024-04-04:
Btw, I didn't mention it, but I tested with the `keyword` locally before on 3.22.0 and it worked.

jeremiah-c-leary, 2024-04-05:
Evening @avelure ,

> I had a look at the diff and it looks ok, thanks :)

Awesome I will get this merged to master.

--Jeremy

PR #1164 (2024-04-05): Issue#1163:  Fixing documentation on use clause indenting configuration.
Resolves #1163
closed 2024-04-05 by jeremiah-c-leary