jeremiah-c-leary/vhdl-style-guide #1166: --vsg_off length_003 does not disable the rule

willkamp, 2024-04-15:
**Environment**
VHDL Style Guide (VSG) version: 3.23.0

**Describe the bug**
using the code tag `--vsg_off length_003` does not disable rule length_003 as expected.

**To Reproduce**
Run vsg on a file (length_003.vhd) containing

```
architecture rtl of length_003 is

begin

  -- vsg_off length_003 length_001 : successfully disables length_001, but not length_003.
  p_length_003 : process is
  begin

    this_is_a_really_long_line_that_exceeds_length_001_criteria <= and_something_really_long_to_assign_to_it_so_it_exceeds_120_columns;

    a <= b; -- repeat this line 500 times to exceed default max length for rule length_003.

  end process p_length_003;

end architecture rtl;
```

Get this output:
```
$ vsg length_003.vhd 
================================================================================
File:  length_003.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 736
Total Violations:    1
  Error   :     0
  Warning :     1
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  length_003                | Warning    |          6 | Reduce process to less than 500 lines
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
Get a violation for length_003 - which should have been disabled.
Length_001 violation was disabled. So `-- vsg_off` code tag is being recognised.

Same behaviour with just a plain code tag `-- vsg_off`

**Expected behavior**
Expect zero violations - since length_003 and length_001 have both been disabled.

jeremiah-c-leary, 2024-04-16:
Morning @willkamp,

I pushed an update to the `issue-1166` branch.  When you get a chance could you check it on your end and let me know if it is working?

Thanks,

--Jeremy

willkamp, 2024-04-16:
Hi @jeremiah-c-leary ,
Thanks for the quick fix :)
I can confirm that the branch issue-1164 fixes the problem, both in the above test case and in my actual code.
Cheers,
Will

jeremiah-c-leary, 2024-04-17:
Awesome,  I will get this merged to master.

--Jeremy

closed 2024-04-17 by jeremiah-c-leary