jeremiah-c-leary/vhdl-style-guide #1170: Extended configurability

LarsAsplund, 2024-05-15:
I'm introducing vsg in a project which has a history of mixed styles. Some have used alignment, some have not. Moving forward, we will remove all alignment and use `--fix` to enforce that in git hooks. While doing that, we found the following issues (in order of priority):

1. When running `--fix`, we managed to remove spaces after `=>` in port and generic mappings but not spaces before `=>`. This means that these lines are only partly "de-aligned" after `--fix`. For example, `my_port       => some_signal,`.
2. If we have a hierarchical reference (external name), for example `<< signal dut.foo       : signal >>`, we have not been able to remove the spaces before `: `.
3. This is more an indentation issue but if we have a long condition to an (els)if statement and break the line into several, it would be nice if the second line could be indented. Like this:
``` vhdl
elsif condition_a and
  condition_b then
```

jeremiah-c-leary, 2024-05-16:
Afternoon @LarsAsplund 

I am currently away from a computer for the next three weeks, but I can answer some of these questions.

1.  It appears there is no rule to force spacing before the assignment operator in port and generic maps.  This is a little puzzling to me as I would expect one to exist.  I will work on adding a rule when I return sometime after June 2nd.

2.  Have you checked the configuration of rule signal_006?  There is an option to set the spacing to 1.  The default is at least one space to allow for alignment.

3.  Have you checked the configuration of rule if_009?  It handles alignment of multiline conditions in if expressions.  If you disabled all the alignment rules using via a group, You can enable that rule invidually.

Regards,

--Jeremy

ru551n, 2024-05-17:
I'll respond here since I am the "active" user of this here at the moment.

> 1. It appears there is no rule to force spacing before the assignment operator in port and generic maps.  This is a little puzzling to me as I would expect one to exist.  I will work on adding a rule when I return sometime after June 2nd.
Appreciate it! :)

> 2. Have you checked the configuration of rule signal_006?  There is an option to set the spacing to 1.  The default is at least one space to allow for alignment.

We have that one enabled, it doesn't seem to fix the colon around the signal keyword when used in hierarchical references.

> 3. Have you checked the configuration of rule if_009?  It handles alignment of multiline conditions in if expressions.  If you disabled all the alignment rules using via a group, You can enable that rule invidually.

We have that rule enabled, with `align_left: yes` and `align_paren: no`, it still seems to put the next row of statements on the same level.

jeremiah-c-leary, 2024-05-17:
Morning @ru551n ,

> We have that one enabled, it doesn't seem to fix the colon around the signal keyword when used in hierarchical references

I reviewed the rule and it should work with hierarchical references.  Could you post the output of the following command:

vsg -rc signal_006

 > We have that rule enabled, with align_left: yes and align_paren: no, it still seems to put the next row of statements on the same level

Maybe I misunderstood the original question.  Is the issue the indenting is not working or the structure of the expression (adding or removing carriage returns) the issue?

--Jeremy

ru551n, 2024-05-20:
> I reviewed the rule and it should work with hierarchical references. Could you post the output of the following command:

`
vsg -rc signal_006 -c .vsg.yaml
{
  "rule": {
    "signal_006": {
      "indent_style": "spaces",
      "indent_size": 2,
      "phase": 2,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "number_of_spaces": 1
    }
  }
}`

> Maybe I misunderstood the original question. Is the issue the indenting is not working or the structure of the expression (adding or removing carriage returns) the issue?

The issue is that the next level of indentation (after CR) is at the same level as the previous row, for the next row we would like:

`
if cond_a and
  cond_b then
`
Instead of
`
if cond_a and
cond_b then
`

ru551n, 2024-05-20:
> I reviewed the rule and it should work with hierarchical references. Could you post the output of the following command:

```
λ vsg -rc signal_006 -c .vsg.yaml
{                                
  "rule": {                      
    "signal_006": {              
      "indent_style": "spaces",  
      "indent_size": 2,          
      "phase": 2,                
      "disable": false,          
      "fixable": true,           
      "severity": "Error",       
      "number_of_spaces": 1      
    }                            
  }                              
}                                
```

jeremiah-c-leary, 2024-05-21:
Afternoon @ru551n ,

It appears the rule is configured correctly for a single space.  I will need to get to a computer before I can debug any further.  I will be back from vacation on June 2nd and will take a look at these issues shortly after.

Regards

--Jeremy

jeremiah-c-leary, 2024-06-02:
Morning @ru551n ,

I have added two new rules to enforce whitespace before port map and generic map assigment operators to branch `issue-1170`.   This should address the first issue.  When you get a chance could you check it out on your end and see if it is working for you.

For the second issue, I have the following code snippet:
```vhdl
  1 architecture rtl of fifo is
  2
  3   signal dut.foo      : std_logic;
  4
  5 begin
  6
  7 end architecture rtl;
```
I used the following configuration file:
```json
  1 "rule": {
  2   "global": {
  3      "disable" : true
  4   },
  5   "signal_006": {
  6      "disable" : false,
  7      "number_of_spaces" : 1
  8   }
  9 }
```
when I run vsg against the file using the configuration I get the following code:
```vhdl
  1 architecture rtl of fifo is
  2
  3   signal dut.foo : std_logic;
  4
  5 begin
  6
  7 end architecture rtl;
```
I must be missing something as rule `signal_006` seems to be enforcing a single space before the colon.  Could you post a snippet that shows the issue you are seeing?

Regards,

--Jeremy

jeremiah-c-leary, 2024-06-02:
Afternoon @ru551n ,

Is there a chance you could post a snippet of the third issue?  I have my test example and it seems to be working as expected:
```vhdl
  1 architecture rtl of fifo is
  2
  3   signal dut.foo      : std_logic;
  4
  5 begin
  6
  7   PROC_1:process
  8
  9   begin
 10
 11     if condition_a and condition_b then
 12     elsif condition_a and
 13                         condition_b then
 14     end if;
 15
 16   end process PROC_1;
 17
 18 end architecture rtl;
```
with this configuration:
```json
  1 "rule": {
  2   "signal_006": {
  3      "disable" : false,
  4      "number_of_spaces" : 1
  5   },
  6   "if_002": {
  7     "parenthesis": "remove"
  8   },
  9   "if_009": {
 10     "align_left": true,
 11     "align_paren": false
 12   }
 13 }
```
yields:
```vhdl
  1 architecture rtl of fifo is
  2
  3   signal dut.foo : std_logic;
  4
  5 begin
  6
  7   proc_1 : process is
  8   begin
  9
 10     if condition_a and condition_b then
 11     elsif condition_a and
 12       condition_b then
 13     end if;
 14
 15   end process proc_1;
 16
 17 end architecture rtl;
```
I feel I am missing something on the last issue.

--Jeremy

jeremiah-c-leary, 2024-06-16:
Morning @ru551n and @LarsAsplund ,

Just wanted to ping you on this issue to see if we can move this forward.

Thanks,

--Jeremy

ru551n, 2024-06-17:
> Morning @ru551n and @LarsAsplund ,
> 
> Just wanted to ping you on this issue to see if we can move this forward.
> 
> Thanks,
> 
> --Jeremy

Due to some more urgent stuff being requested this has been on hold for a while. Now I am back on it though, so here goes:

1. The two new rules work perfectly, cheers!
2. Aliases as hierarchical references still doesn't work properly, i.e
```vhdl
alias test is << signal dut.test :     std_logic >>;
``` 
Does not get aligned.
3. You are correct, it does indeed work properly.

jeremiah-c-leary, 2024-06-19:
Evening @LarsAsplund ,

So apparently it took me awhile to figure out you were referring to `external_signal_name`s.  I updated the parser to detect and parse `external_signal_name`, `external_variable_name`, and `external_constant_name`.  I also added rules for `external_signal_name`.  There are several whitespace and a case rule for `signal`.  Could you give this update a try and let me know if it fixes the whitespace issue.

Also, would you like similar rules for `external_variable_name` and `external_constant_name`?

Regards,

--Jeremy

ru551n, 2024-06-19:
> Evening @LarsAsplund ,
>
> So apparently it took me awhile to figure out you were referring to `external_signal_name`s. I updated the parser to detect and parse `external_signal_name`, [...]

Works nicely!

Regarding external constants and variables, it's not something we use commonly, but if it is not too much work it is better to have than not.

Thank you!

-- Sebastian

LarsAsplund, 2024-06-19:
@jeremiah-c-leary It works perfectly on external names when used with aliases but not on external names used in a context like this:

`probe_signal <= <<    some.hierarchy : std_logic   >>`

We use both styles

jeremiah-c-leary, 2024-06-22:
Evening @LarsAsplund ,

I added rules for the `external_constant_name`s and `external_variable_name`s.  

Regards,

--Jeremy

jeremiah-c-leary, 2024-07-21:
Evening @ru551n and @LarsAsplund ,

Just wanted to ping you on this issue to see if we can move it forward to completion.

Thanks,

--Jeremy

LarsAsplund, 2024-08-26:
@jeremiah-c-leary Sorry for the delay. 

```
probe_signal <= <<    some.hierarchy : std_logic   >>
```

is a typo from my side. I think this issue can be closed but I will make a test run with the latest version to see if I can find any remaining issues.

LarsAsplund, 2024-08-27:
One issue I see is if the external name references a signal in a path generated by a generate statement. In those cases the external name will start with `some.path.generate_label(i)` where `i` is a number. VSG doesn't like the parenthesis in the the path.

LarsAsplund, 2024-08-27:
Other than that I can't see any issues on our code based. If that is fixed, these changes can be merged and released as far as I'm concerned.

jeremiah-c-leary, 2024-09-09:
Evening @LarsAsplund,

I will take a look at the condition with the parenthesis.

--Jeremy

jeremiah-c-leary, 2024-10-02:
Evening @LarsAsplund ,

I have pushed an update to handle parenthesis.  When you get a chance could you check it out.

Thanks,

--Jeremy

LarsAsplund, 2024-10-02:
@jeremiah-c-leary Looks good to me! Will you make a release?

jeremiah-c-leary, 2024-10-03:
Evening @LarsAsplund,

I will merge this to master and schedule it for release this weekend.

--Jeremy

PR #1173 (2024-10-03): Issue#1170:  Adding rules to enforce spaces before generic map and port map assignment operators.
Resolves #1170
closed 2024-10-03 by jeremiah-c-leary