jeremiah-c-leary/vhdl-style-guide #1171: Rule for "space around operators" (?)

ungultig1, 2024-05-17:
Hi there,

i couldn't find the following rule yet, so could be included in future releases.

I'd like to have a format rule for this case, where it inserts a single space before and after operators:

violation:

```
if (readAddr>=writeAddr) then
 ...
end if;
```

Fix:

```
if (readAddr >= writeAddr) then
 ...
end if;
```

Thank you.

Greetings,

jeremiah-c-leary, 2024-05-17:
Evening @ungultig1 ,

You are correct, there is no rule for whitespace around most operators.  I am currently away from a computer until June 2nd.  I will add a rule shortly after that date.

Regards,

--Jeremy

jeremiah-c-leary, 2024-06-02:
Morning @ungultig1 ,

I have pushed an update for this issue to the `issue-1171` branch.  When you get a chance could you validate it on your end and let me know if it is working on your end?

Thanks,

--Jeremy

jeremiah-c-leary, 2024-06-16:
Morning @ungultig1 ,

Just wanted to ping you to see if you had a chance to check out the update.

Thanks,

--Jeremy

domWalters, 2024-07-03:
@jeremiah-c-leary I'm currently getting your tool working in my workflow and was looking for exactly this rule.

Hopefully I'll be able to test this at some point tomorrow.

Leaving this comment to hopefully remind me 😊

jeremiah-c-leary, 2024-07-09:
Evening @ungultig1 ,

Just wanted to ping you to see if you had time to validate the update.

--Jeremy

domWalters, 2024-07-09:
> Morning @domWalters ,
> 
> It would be great if you could validate this issue.
> 
> --Jeremy

Tried this new config option this morning, looks good to me.

jeremiah-c-leary, 2024-07-10:
Awesome,

I will merge this to master.

--Jeremy

PR #1172 (2024-07-10): Issue#1171:  Adding rule whitespace_100 to enforce single spacing around relational operators.
Resolves #1171
closed 2024-07-10 by jeremiah-c-leary