jeremiah-c-leary/vhdl-style-guide #1176: Rule for new lines between end of generic map and port map

tossegus, 2024-06-13:
Hi!

I tried looking for rules regarding the end of the `generic map` in combination with the start of the `port map`, but couldn't find any rule that matches what I was looking for (see below).

Violation:
```
generic map (
...
)
port map (
...
);
```

Fix:
```
generic map (
...
) port map (
...
);
```

**Describe alternatives you've considered**
Locally in my project I added a number of consecutive greps to find these violations (which works well), but having the rule directly in VSG would be very much preferred 😄

jeremiah-c-leary, 2024-06-14:
Evening @tossegus ,

I would agree there is no rule to enforce the behavior you are looking for.  The rule `instantiation_005` is close, but it will always want to put `port map` on it's own line.   I am guessing you would want `port map` to be moved to it's own line if it was on the same line as the instance declaration?

Violation
```vhdl
  U_FIFO : FIFO port map (

  U_FIFO : FIFO
     generic map (
     )
     port map (
```
Fix
```vhdl
  U_FIFO : FIFO
     port map (

  U_FIFO : FIFO
     generic map (
     ) port map (
```
If so I could add to configuration options with the following defaults:
```yaml
rules:
   instantiation_005:
       after__instantiation:  "new line"
       after_generic_map: "new line"
```
This would match the current behavior.  Then for your formatting the configuration would be:
```yaml
rules:
   instantiation_005:
       after_instantiation:  "new line"
       after_generic_map: "no new line"
```
Would that update give you the desired formatting?

--Jeremy

tossegus, 2024-06-14:
> Evening @tossegus ,
>
> I would agree there is no rule to enforce the behavior you are looking for. The rule `instantiation_005` is close, but it will always want to put `port map` on it's own line. I [...]

Hi!

Yes, that would give me the wanted formatting. 🙂 

/  Tossegus

jeremiah-c-leary, 2024-06-16:
Morning @tossegus ,

I pushed an update for this to the `issue-1176` branch.  There are two new configuration options that will allow you to add or remove  carriage returns depending on what comes before the port map aspect.  When you get a chance, could you check it out on your end and let me know if it works for you?

Thanks,

--Jeremy

tossegus, 2024-06-19:
Good morning!

It looks like it works on my side! 🙂 

/ tossegus

jeremiah-c-leary, 2024-06-19:
Awesome,

I will get this merged to master.

--Jeremy

PR #1180 (2024-06-19): Issue#1176:  Updating rule instantiation_005 to allow for adding or removing carriage returns
Resolves #1176