jeremiah-c-leary/vhdl-style-guide #1177: Prefix/suffix exceptions not exempted from case checking in instantiation_008

maltaisn, 2024-06-13:
Similar to #644, an error is issued when prefix/suffix exceptions don't match configured case for `instantiation_008`, and maybe other rules which I haven't tested.

Example configuration:
```yml
rule:
  group:
    case:
      severity: Warning
  instantiation_008:
    case: lower
    prefix_exceptions:
      - "P_"
    suffix_exceptions:
      - "_S"
```
With file:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  port (
    a : in    std_logic;
    b : in    std_logic
  );
end entity test;

architecture behavioral of test is

  component comp is
    port (
      a : in    std_logic;
      b : in    std_logic
    );
  end component;

begin

  P_comp : component comp
    port map (
      a => a,
      b => b
    );

  comp_S : component comp
    port map (
      a => a,
      b => b
    );

end architecture behavioral;
```

Gives output:
```text
Phase 7 of 7... Reporting
Total Rules Checked: 738
Total Violations:    2
  Error   :     0
  Warning :     2
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  instantiation_008         | Warning    |         22 | Change "P_comp" to "p_comp"
  instantiation_008         | Warning    |         28 | Change "comp_S" to "comp_s"
----------------------------+------------+------------+--------------------------------------
```
The exceptions should be completely exempt from case checking. I'm using v3.24.0.

jeremiah-c-leary, 2024-06-14:
Evening @maltaisn ,

The prefix and suffix exceptions are applied at the same time.  So the configuration you have would be checking the following:
```vhdl
  P_comp_S : component comp
    port map (
      a => a,
      b => b
    );
```
Since your first instantiation is missing the suffix `_S` it is considered a violation.
The second instance is missing the prefix `_P` so it is also considered a violation.

I scanned the documentation and there is nothing that covers using both prefix and suffix exceptions at the same time.

The question is, how would you like to proceed?

--Jeremy

maltaisn, 2024-06-14:
I would have expected the check to succeed if either a prefix is present, a suffix, or one of both.

Benito-E, 2024-06-14:
@maltaisn This is a comment from an outside of what you're working on, but I believe it wouldn't be impossible to create a rule that enforces one of both. There are two base rules (rules that are the foundation for many of the full set of VSG's built in rules) that manage prefix and suffix rules: vsg.rules.token_prefix.py and vsg.rules.token_suffix.py. While the standard way to create a rule usually involves extending your rule off of one of these base rules, in order to capture the functionality of both of these rules, you could create a unique rule that instantiates both these rules as objects, runs their analysis manually, and adds violations based on the violations present within those objects.

--Benito

jeremiah-c-leary, 2024-06-16:
Evening @maltaisn and @Benito-E ,

I reviewed the implementation and I believe the best course of action would be to update the prefix/suffix base rule to check for all permutations of prefix and suffix exceptions if both are defined.   I should have considered those permutations when I first added the exceptions.

I have pushed an update for this to `issue-1177` branch.  When you get a chance could you check it out on your end and let me know if it is working for you?

Thanks,

--Jeremy

maltaisn, 2024-06-17:
It works for me, thanks!

closed 2024-06-17 by maltaisn

jeremiah-c-leary, 2024-06-18:
Awesome I will merge this to master.

PR #1181 (2024-06-18): Issue#1177: Updating prefix and suffix rules to check for prefix or suffix if both exceptions are provided.
Resolves #1177