jeremiah-c-leary/vhdl-style-guide #1178: Rule concerning label presence after end keyword on component declaration

maltaisn, 2024-06-13:
There doesn't appear to be a rule to add/remove the label after the `end` keyword on a component declaration, similar to process_018, package_014, architecture_024, etc.

**Violation**
```vhdl
end component;
```
**Fix**
```vhdl
end component comp;
```

jeremiah-c-leary, 2024-06-14:
Evening @maltaisn ,

> There doesn't appear to be a rule to add/remove the label after the end keyword on a component declaration, similar to process_018, package_014, architecture_024, etc.

You are correct.  Not sure how I missed that one.  I will add a rule for it.

--Jeremy

jeremiah-c-leary, 2024-06-14:
Morning @maltaisn ,

I pushed an update for this to the `issue-1178` branch.  When you get a chance could you check it out on your end and let me know if it is working for you?

Thanks,

--Jeremy

maltaisn, 2024-06-14:
Thanks it works.

closed 2024-06-14 by maltaisn

jeremiah-c-leary, 2024-06-14:
Awesome, I will merge this to master.

--Jeremy

PR #1179 (2024-06-14): Issue#1178:  Adding rule component_022 to add or remove optional component_simple_name.
Resolves #1178