jeremiah-c-leary/vhdl-style-guide #1186: Forcing single space when disabling alignment rules

maltaisn, 2024-06-26:
Alignment can be interesting to have for entity declarations, signals, etc., but I prefer to leave it aside for a few things, like component declarations, assignments, etc., to improve version control diffs. As such, I disabled a few alignment rules.

However, while formatting a whole repository, I found many cases where an attempt an alignment was made in the past, but not kept up-to-date with new changes. I would like to normalize these cases by removing extra spaces and leaving only one, but it seems there's no option to do it.

For example:
```vhdl
component Comp is
    port (
        a   : in std_logic;
        aa  : out std_logic;
        aaa : in std_logic;
        aaaaa   : out std_logic;
        aaaaaaa : out std_logic;

        b                       : in std_logic;
        bb : out std_logic
    );
end component;
```
Would become:
```vhdl
component Comp is
    port (
        a : in std_logic;
        aa : out std_logic;
        aaa : in std_logic;
        aaaaa : out std_logic;
        aaaaaaa : out std_logic;

        b : in std_logic;
        bb : out std_logic
    );
end component;
```

I'm not sure what would be the best way to describe this in the configuration since there are many different cases for alignment. In a way, this means devolving alignment rules to whitespace rules with a number of spaces setting.

jeremiah-c-leary, 2024-06-27:
Evening @maltaisn ,

There is typically a rule for whitespace that corresponds to an alignment rule.  To enforce a single space before the colon, change the `number_of_spaces` option on rule `port_020` to `1`.

```json
  "rule": {
    "port_020": {
       "number_of_spaces" : 1
    }
  }
```

Regards,

--Jeremy

maltaisn, 2024-06-27:
I didn't know about this rule. However I haven't found equivalent rules for those I have disabled: 

- `architecture_029`: alignment of declaration names after keyword.
- `component_017`: alignment of colon in component port declaration (`port_020` applies equally to entity and component, I would only like to have them aligned in entity as they are inspected more often than components which are simply boilerplate).
- `declarative_part_400`: alignment of assignments in declarative part.
- `entity_018`: alignment of default assignments in entity port/generic map.
- `instantiation_010`: alignment of port/generic map.
- `process_400`: alignment of assignments within a process body.

jeremiah-c-leary, 2024-06-28:
Evening @maltaisn ,

Whitespace rules are assigned to the lowest production in the LRM as possible.  In most of the cases you listed, the alignment rule is at a "higher" production than the whitespace rule.  [This section](https://vhdl-style-guide.readthedocs.io/en/stable/theory_of_operation/rule_selection.html) of the documentation covers the thoughts behind this method.  For example:

### architecture_029

The rules for whitespace before the identifiers are in rules `signal_100`, `file_100`, `variable_100` and `type_100`.

### declarative_part_400

The rules for whitespace before default assignment tokens are in rules `signal_101` and `constant_006`.  There is no rule for variables for shared variables.  VSG's opinion is to remove default assignments on variables using rule `variable_007`

### entity_018

There is rule `generic_014`.  No rule exists for ports though.  VSG's opinion is to remove default assignments on ports using rule `port_012`

### instantiation_010

There are no rules to handle the whitespace before the assignment in port maps or generic maps.

### process_400

There is a rule `sequential_003` which covers whitespace before the assignment operator.

### component_017

The assumption was port clauses would be formatted the same regardless of where they are located.  I will have to think about a way forward on this.

I will add rules to cover whitespace before assignments in port maps and generic maps.

If you could configure the other rules I have identified and let me know how they work out for you or if I am missing anything.

--Jeremy

jeremiah-c-leary, 2024-07-05:
Afternoon @maltaisn ,

I have added more whitespace rules:

= architecture_029
No rules expected to be added

= declarative_part_400
variable_100
variable_101
variable_102

= entity_018
port_100
port_101

= instantiation_010
port_map_100
generic_map_100

= process_400
No rules expected to be added

= component_017
Still thinking about this one

Could check out the rules I have implemented on the `issue-1186` branch and let me know how they are working for you?

Thanks,

--Jeremy

maltaisn, 2024-07-11:
Sorry for the delay. I just tried the new rules and everything works as I wanted.

It's fine for `component_017`, I can live with it and I understand it doesn't make much sense to have them formatted differently.

closed 2024-07-11 by maltaisn

jeremiah-c-leary, 2024-07-12:
Morning @maltaisn ,

> I just tried the new rules and everything works as I wanted.

Awesome, I will get this merged to master.

> It's fine for component_017, I can live with it and I understand it doesn't make much sense to have them formatted differently.

Thanks for understanding.  I have been struggling with the implications of allowing different formatting depending on where a production is located in the code.

Regards,

--Jeremy