jeremiah-c-leary/vhdl-style-guide #1187: record element is incorrectly recognized as type element

mattdominik, 2024-07-01:
**Environment**
Windows 10
VHDL Style Guide (VSG) version: 3.24.0

**Describe the bug**
Record element is incorrectly recognized as type element. If a record element has the same name as a type element, a type_501 error is incorrectly detected.

**To Reproduce**
Steps to reproduce the behavior:

```
library ieee;
use ieee.std_logic_1164.all;

package example_pkg is
    type t_example_in is record
        start : std_logic;
    end record;

    type t_example_out is record
        done : std_logic;
    end record;
end package;

library ieee;
use ieee.std_logic_1164.all;

library work;
use work.example_pkg.all;

entity example is
    port(
        i_clk   : in  std_logic;
        i_reset : in  std_logic;
        d       : in  t_example_in;
        q       : out t_example_out
    );
end;

architecture rtl of example is

    type t_state is (BIT_LENGTH, MUL, DATA_STREAM_LENGTH);

    type t_reg is record
        state              : t_state;
        data_stream_length : std_logic_vector(21 downto 0);
        done               : std_logic;
    end record;

    constant C_REG_T_INIT : t_reg := (
        state              => BIT_LENGTH,
        data_stream_length => (others => '0'),
        done               => '0'
    );

    signal r_next, r : t_reg;

begin

    -------------------------------------------------------------------------------------
    -- Output Signals
    -------------------------------------------------------------------------------------
    q.done <= r.done;

    -------------------------------------------------------------------------------------
    -- Combinatorial process
    -------------------------------------------------------------------------------------
    comb : process(all)
        variable v : t_reg;
    begin
        v := r;

        v.done := '0';

        case r.state is
            when BIT_LENGTH =>
                if d.start = '1' then
                    v.state := MUL;
                end if;
            when MUL =>
                v.state := DATA_STREAM_LENGTH;

            when DATA_STREAM_LENGTH =>
                v.done  := '1';
                v.state := BIT_LENGTH;
        end case;

        r_next <= v;
    end process;

    regs : process(i_clk)
    begin
        if rising_edge(i_clk) then
            if i_reset = '1' then
                r <= C_REG_T_INIT;
            else
                r <= r_next;
            end if;
        end if;
    end process;
end;

```

Output:
```
================================================================================
File:  C:/example.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 649
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  type_501                  | Error      |         41 | Change data_stream_length to DATA_STREAM_LENGTH
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
No error should be detect

jeremiah-c-leary, 2024-07-11:
Morning @mattdominik,

Just wanted to let you know I am investigating this issue and it will take me awhile to resolve it.

--Jeremy

mattdominik, 2024-08-08:
Any news here yet?

jeremiah-c-leary, 2024-08-17:
Morning @mattdominik ,

Currently this has stalled as I work is demanding more of my time and I am trying to decide if I should switch to rust_hdl as a parser.  I am delaying the decision to write another parser to generate a data structure that I can resolve your issue.  Although it is looking more and more likely that I will need to.

Regards,

--Jeremy

mattdominik, 2024-11-26:
Morning @jeremiah-c-leary,

Are there any activities planned here?

jeremiah-c-leary, 2024-12-09:
Evening @mattdominik ,

I am currently mulling over the updates I need to the data structure to support this.  I would not expect anything this year.

Thanks for pinging me.

--Jeremy

mattdominik, 2025-02-21:
hi @jeremiah-c-leary,

Wanted to quietly ask again if there is any news?

jeremiah-c-leary, 2025-03-01:
Evening @mattdominik ,

I have started the prep work by starting issue #881.

--Jeremy

mattdominik, 2025-05-26:
Are there any updates?