jeremiah-c-leary/vhdl-style-guide #1188: Comment tricks vsg into incorrect parsing state

domWalters, 2024-07-03:
**Environment**
Ubuntu 20.04
VSG 3.23.0

**Describe the bug**
Certain comments can cause the vsg parser to think it is parsing statements that it isn't.

In the example below, vsg thinks it is interpreting a procedure call statement, when it is actually an if else.

**To Reproduce**
Run `vsg` against a file with this content:

```vhdl
-- [*a*/D]
library ieee;
use ieee.std_logic_1164.all;

entity uut is
    port (
        a : in  boolean;
        b : out std_logic
    );
end entity;

architecture rtl of uut is
begin

    process
    begin
        if a then
            b <= '0';
        else
            b <= '1';
        end if;
    end process;

end architecture;
```

FYI: This is a drastically cut down and modified version of a file generated by Xilinx Vivado.
The comment that has been reduced to `[*a*/D]`, is actually an XDC `set_false_path` constraint that looks up a pin using the `*` glob.

**Expected behavior**
`vsg` should complete parsing this file.

**Screenshots**
When the above is run through `vsg`:
```console
> vsg error.vhd

Error: Unexpected token detected while parsing procedure_call_statement @ Line 19, Column 9 in file error.vhd
       Expecting : ;
       Found     : else
```

This error points to the first character of the word `else`.

jeremiah-c-leary, 2024-07-05:
Morning @domWalters ,

I have pushed a fix to the `issue-1188` branch.  It turns out the `*/` was giving the comment parser problems.  When you get a chance could you run this branch on your end and let me know if it fixes the issue?

Thanks,

--Jeremy

domWalters, 2024-07-09:
> Morning @domWalters ,
> 
> I have pushed a fix to the `issue-1188` branch. It turns out the `*/` was giving the comment parser problems. When you get a chance could you run this branch on your end and let me know if it fixes the issue?
> 
> Thanks,
> 
> --Jeremy

Confirmed working, thanks!

jeremiah-c-leary, 2024-07-10:
Awesome, I will get this merged to master.

--Jeremy

PR #1191 (2024-07-10): Issue#1188:  single comments with */ would be incorrectly classified and fail to parse.
Resolves #1188
closed 2024-07-10 by jeremiah-c-leary