jeremiah-c-leary/vhdl-style-guide #1189: Better diagnostics when local indent rules don't match default config

abyszuk, 2024-07-04:
Hello,

I have my local, complete configuration (including both `indent` and `rule` groups).
The problem is that often when I update VSG the upstream `indent` config changes in a way that's not compatible with my local config resulting in the following error:
```
Traceback (most recent call last):
  File "/home/adrian/.local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/__main__.py", line 111, in main
    oConfig = config.New(commandLineArguments)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/config.py", line 314, in New
    dIndent = read_indent_configuration(dConfig)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/config.py", line 211, in read_indent_configuration
    dReturn["indent"]["tokens"][sGroup][sToken][sParameter] = dGroups[sGroup][sToken][sParameter]
KeyError: 'target'
```
I understand that I have to update my local config, but the problem it's time consuming because I have to manually compare the configs (or look at the VSG git diff).
It would be great if VSG could print at least the name of problematic Group and/or Token.
It would be even better if VSG could print a detailed description, for example:
```
Configuration mismatch, invalid key in local config.
Local config:
    simple_variable_assignment:
      target:
        after: current
        token: current

Upstream config
    simple_variable_assignment:
      aggregate_close_parenthesis:
        after: '-1'
        token: '-1'
      aggregate_open_parenthesis:
        after: '+1'
        token: 'current'
      simple_name:
        after: 'current'
        token: 'current'
```

jeremiah-c-leary, 2024-07-04:
Morning @abyszuk ,

I suppose I could use a `try...except` on a `KeyError`.  Then in the except, figure out which key is incorrect and report back the valid keys.

We can work through the details as I implement an update.

--Jeremy

jeremiah-c-leary, 2024-07-06:
Morning @abyszuk ,

I pushed an update for this issue to the `issue-1189` branch.  When you get a chance could you give it a try and let me know what you think?

Thanks,

--Jeremy

abyszuk, 2024-07-08:
Thank you @jeremiah-c-leary , that looks perfect!

jeremiah-c-leary, 2024-07-09:
Awesome,  I will get this merged to master.

--Jeremy

PR #1194 (2024-07-09): Issue#1189:  Adding error reporting for invalid indent configurations.
Resolves #1189
closed 2024-07-09 by jeremiah-c-leary