jeremiah-c-leary/vhdl-style-guide #1190: Whitespace rules should allow <= operator

abyszuk, 2024-07-05:
Currently VSG allows only to configure either a constant or minimum number of spaces.
https://vhdl-style-guide.readthedocs.io/en/3.24.0/configuring_whitespace_rules.html#configuring-whitespace-rules

There are cases where a maximum number of spaces would be useful as well, i.e. `number_of_spaces: <=1`.
For example rule [process_024](https://vhdl-style-guide.readthedocs.io/en/3.24.0/process_rules.html#process-024)

I'm fine with code having 0 or 1 spaces:
```vhdl
cic_proc_p: process(all)

cic_proc_p : process(all)
```
But for sure I wouldn't want the code like this:
```vhdl
cic_proc_p           : process(all)
```

jeremiah-c-leary, 2024-07-05:
Morning @abyszuk,

I imagine the fix would be the maximum allowed in a `<=`?

For example the following code snippet:
```
cic_proc_p        : process(all)
```
if the whitespace rule was configured for `number_of_spaces: <= 2` the result would be:
```
cic_proc_p  : process(all)
```
two spaces before the colon?

--Jeremy

abyszuk, 2024-07-05:
Yes, that's my thought as well.

jeremiah-c-leary, 2024-07-05:
Afternoon @abyszuk ,

I pushed an update for this to the `issue-1190` branch.  When you get a chance could you check it out on your end and let me know if it works for you?

Thanks,

--Jeremy

abyszuk, 2024-07-08:
Hello @jeremiah-c-leary 
I've tested the fix on my end and I can confirm it works as intended.
Thank you for the fast update!

jeremiah-c-leary, 2024-07-09:
Evening @abyszuk ,

I pushed an update to the no spaces case.  Essentially if there are no spaces then the `<=1` should always pass.  Let me know if there are any other issues.

Thanks,

--Jeremy

abyszuk, 2024-07-09:
Yes, I can confirm this works properly now.
Again, thank you for the fast update and great work!

jeremiah-c-leary, 2024-07-09:
Awesome, I will get this merged to master.

> Again, thank you for the fast update and great work!

Thanks for the kind words.

--Jeremy

PR #1193 (2024-07-09): Issue#1190:  Adding <= and < options to whitespace rules.
Resolves #1190
closed 2024-07-09 by jeremiah-c-leary