jeremiah-c-leary/vhdl-style-guide #1195: Ignore Components for port rules

el-coder-sb, 2024-07-09:
**Is your feature request related to a problem? Please describe.**
When running vsg on old projects, because of binded (also old coded) components we can´t achieve a warning free vsg run on a higher entity without changing all lower entities. Which often results in not changing anything because of the amount of work :-(

**Describe the solution you'd like**
It would be great to configure if port rules should be applied to component declarations.

I don´t know if there are other rules which also should be considered. I think mainly the naming rules are the ones which should be configurable.

**Describe alternatives you've considered**

Example: A suffix _O oder _I is required for ports. 
```
rule:
  port_025:
    disable: false
    suffixes: ['_I', '_O']
```
The following code should throw only warnings for the top entity port names: 

--

**Additional context**
```vhdl
library std;
library ieee;
    use std.standard.all;
    use ieee.std_logic_1164.all;

entity top_entity is
    port(
        RST_N            : in  std_logic;
        CLK              : in  std_logic
    );
end;

architecture top_entity_arc of top_entity is

    component lower_entity
        port (
            RST_N     : in  std_logic;
            CLK       : in  std_logic
        );
    end component;

begin

end top_entity_arc;
```

jeremiah-c-leary, 2024-07-12:
Morning @el-coder-sb ,

Have you considered using an embedded code tag to disable the rule?

For example:
```vhdl
library std;
library ieee;
    use std.standard.all;
    use ieee.std_logic_1164.all;

entity top_entity is
    port(
        RST_N            : in  std_logic;
        CLK              : in  std_logic
    );
end;

architecture top_entity_arc of top_entity is

    -- vsg_off port_025
    component lower_entity
        port (
            RST_N     : in  std_logic;
            CLK       : in  std_logic
        );
    end component;
    -- vsg_on

begin

end top_entity_arc;
```
A code tag would allow you to be precise on where a rule is applied.

--Jeremy

el-coder-sb, 2024-07-12:
@jeremiah-c-leary Oh thanks for that hint. Of course this would be a valid workaround. Up to now we never used tags, but you are right, this would be a solution. I will do some tests on some of our old files.

jeremiah-c-leary, 2024-07-12:
Unfortunately code tags pollute your code, but if you do not have very many of them it seems to be an acceptable trade off.

el-coder-sb, 2024-07-12:
Normally all component declarations go together so we need that tag only ones. I´m fine with this solution.

closed 2024-07-12 by el-coder-sb