jeremiah-c-leary/vhdl-style-guide #1196: Function/Procedure name as optional in end statement

domWalters, 2024-07-09:
**Is your feature request related to a problem? Please describe.**

There are various existing rules to remove "names" from `end` statements. There isn't one for `function`s or `procedure`s as far as I can tell.

**Describe the solution you'd like**

The following is valid VHDL:
```vhdl
function test return boolean is
begin
  return true;
end function;
```

I would like to enforce the removal of any function name in the `end function` statement. The same could be said for the procedure statement.

**Describe alternatives you've considered**

Within `vsg`, I'm unaware of an alternative.

jeremiah-c-leary, 2024-07-11:
Morning @domWalters,

I pushed an update for this to the `issue-1196` branch.  Two rules were added:  `procedure_014` and `function_020`.  When you get a chance could you check it on your end and let me know if it works for you?

Thanks,

--Jeremy

domWalters, 2024-07-11:
> Morning @domWalters,
> 
> I pushed an update for this to the `issue-1196` branch. Two rules were added: `procedure_014` and `function_020`. When you get a chance could you check it on your end and let me know if it works for you?
> 
> Thanks,
> 
> --Jeremy

Hey,

Ran `vsg` over one existing repository and it did what I expected for `function`. No `procedure`s in that repo, so I'll try some others and get back to you.

domWalters, 2024-07-11:
> Hey,
>
> Ran `vsg` over one existing repository and it did what I expected for `function`. No `procedure`s in that repo, so I'll try some others and get back to you.

Ran over some `procedure`s, looks okay there as well.

jeremiah-c-leary, 2024-07-12:
Awesome, I will get this merged to master.

--Jeremy

closed 2024-07-12 by jeremiah-c-leary

patrick-studer, 2024-07-21:
Hi @jeremiah-c-leary 

I guess your [3.25.0: Announcements](https://github.com/jeremiah-c-leary/vhdl-style-guide/releases/tag/3.25.0) include wrong information (typo).

Here in this issue, you describe the fix was applied to `procedure_014` and `function_020`. In the "changelog" you wrongly refer to `procedure_020` instead of `procedure_014`.

Would be nice if you could update the "changelog" since it points user to a non-existing rule...

BTW: I'm really happy about your documentation style. The changelog helps to follow changes which are needed to the configuration file. Love it!

jeremiah-c-leary, 2024-07-21:
Evening @patrick-studer ,

> Here in this issue, you describe the fix was applied to procedure_014 and function_020. In the "changelog" you wrongly refer to procedure_020 instead of procedure_014.

Good catch, and I had the opportunity to learn how to update a tag commit message.  If you check the tag message it should be updated correctly.

> BTW: I'm really happy about your documentation style. The changelog helps to follow changes which are needed to the configuration file. Love it!

Thanks for the kind words.  It takes a bit of time to document the release, but it is worth it in the end.

--Jeremy

PR #1197 (2024-07-12): Issue#1196: Adding rules to check for subprogram designators.
Resolves #1196