jeremiah-c-leary/vhdl-style-guide #1198: Regex case rule fails on port_map_002

maltaisn, 2024-07-12:
The regex option fails for `port_map_002` it seems:
```
Traceback (most recent call last):
  File "C:\dev\vhdl-style-guide\bin\vsg", line 16, in <module>
    main()
  File "C:\dev\vhdl-style-guide\vsg\__main__.py", line 154, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "C:\Program Files\Python311\Lib\multiprocessing\pool.py", line 873, in next
    raise value
AttributeError: 'rule_002' object has no attribute 'oRegex'
```
See #1182

jeremiah-c-leary, 2024-07-13:
Evening @maltaisn ,

I pushed an update for this to the `issue-1198` branch.  When you get a chance could you verify it works on your end?

Thanks,

--Jeremy

maltaisn, 2024-07-18:
It works, thanks.

closed 2024-07-18 by maltaisn

jeremiah-c-leary, 2024-07-19:
Awesome,

I will get this merged to master.

--Jeremy

PR #1199 (2024-07-19): Issue#1198:  Adding regex option to rule port_map_002.
Resolves #1198