jeremiah-c-leary/vhdl-style-guide #1201: Unable to have a series of port, port map declarations, signal declarations all be aligned even across comments or white space

greenalden, 2024-07-19:
I have been having trouble setting up a YAML rules file for VSG. I have had no problems doing some things like,

rule :
  port_010 :
    case : 'upper_or_lower'

However, I have been having problems answering what I stated in my questions. My understanding is what I have below should have worked.

rule :
  process_031:
    blank_line_ends_group: 'no'
    comment_line_ends_group : 'no'
  port_010 :
    case : 'upper_or_lower'

Though changing any of these, or to the best I can tell changing any of the process_031 rules, doesn't have any effect. Thank you to anyone who can help with this problem.

jeremiah-c-leary, 2024-07-19:
Afternoon @greenalden,

Could you post an example of your desired formatting.  It would help in determining if the current rule set is sufficient or if I need to  create additional rules.

--Jeremy

greenalden, 2024-07-20:
I am hoping to keep everything in line across spaces and comments for things like this, words like std_logic; should also be in line but I just can't show it well here

    EXAMPLE          : in    std_logic;
    EXAMPLE          : out   std_logic;
    EXAMPLE          : out   std_logic;
    EXAMPLE          : in     std_logic;
    EXAMPLE          : out   std_logic;
    EXAMPLE          : out   std_logic;
    EXAMPLE          : out   std_logic_vector(3 downto 0);

    -- This is a comment
    EXAMPLE          : in    std_logic;
    EXAMPLE          : in    std_logic;

    EXAMPLE          : in    std_logic_vector(3 downto 0);
    EXAMPLE          : in    std_logic_vector(3 downto 0)

and also things like this

      EXAMPLE               => reset,
      EXAMPLE               => reset,
      EXAMPLE               => reset,
      EXAMPLE               => reset,

      EXAMPLE               => open,
      EXAMPLE               => open,

jeremiah-c-leary, 2024-07-21:
Morning @greenalden ,

For the rule `process_031`, I used the following code example:
```vhdl
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6   U_PROC : process is
  7
  8     variable    var1 : boolean;
  9
 10     constant          const1 : integer;
 11
 12     -- Comment
 13     file file1 : load_file_file open read_mode is load_file_name;
 14
 15   begin
 16   end process U_PROC;
 17
 18 end architecture rtl;
 ```
with the the following configuration file:
```yaml
  1 rule:
  2    process_031:
  3       blank_line_ends_group: 'no'
  4       comment_line_ends_group: 'no'
```
and the used the following command:
```mono
vsg -f original.vhd --fix -c config.yaml
```
which resulted in the following formatted VHDL file:
```vhdl
  1
  2 architecture rtl of fifo is
  3
  4 begin
  5
  6   u_proc : process is
  7
  8     variable var1 : boolean;
  9
 10     constant const1 : integer;
 11
 12     -- Comment
 13     file     file1 : load_file_file open read_mode is load_file_name;
 14
 15   begin
 16
 17   end process u_proc;
 18
 19 end architecture rtl;
```
Which shows `var`, `const1`, and `file1` being aligned.

Is there a possibility you could post a small example code snippet and a small configuration test case I can look at?

--Jeremy

greenalden, 2024-07-22:
Here is an example of what I am trying to do.

here is part of the old code with VSG:

  component adc_chan_a_intf is
    port (
      reset                   : in    std_logic;
      adc_clk_p               : in    std_logic;
      adc_clk_n               : in    std_logic;
      adc_data_p              : in    std_logic_vector(k_adc_width - 1 downto 0);
      adc_data_n              : in    std_logic_vector(k_adc_width - 1 downto 0);
      adc_ovr_sync_p          : in    std_logic;
      adc_ovr_sync_n          : in    std_logic;
      fpga_unused_io_1v8_au32 : in    std_logic;

      adc_io_fabric_clk : in    std_logic;
      adc_phy_clk       : out   std_logic;
      adc_data_rd_clk   : out   std_logic;
      adc_data_rdy      : out   std_logic;
      adc_data_out      : out   t_adc_fifo_array;

      pll_clkout0        : out   std_logic;
      pll_clkout1        : out   std_logic;
      pll_locked         : out   std_logic;
      multi_intf_lock    : in    std_logic;
      intf_rdy           : out   std_logic;
      adc_data_overrange : out   std_logic;

      adc_io_vtc_en  : in    std_logic;
      adc_io_vtc_rdy : out   std_logic
    );
  end component adc_chan_a_intf;

  component adc_chan_b_intf is
    port (
      reset                   : in    std_logic;
      adc_clk_p               : in    std_logic;
      adc_clk_n               : in    std_logic;
      adc_data_p              : in    std_logic_vector(k_adc_width - 1 downto 0);
      adc_data_n              : in    std_logic_vector(k_adc_width - 1 downto 0);
      adc_ovr_sync_p          : in    std_logic;
      adc_ovr_sync_n          : in    std_logic;
      fpga_unused_io_1v8_aw33 : in    std_logic;

      adc_io_fabric_clk : in    std_logic;
      adc_phy_clk       : out   std_logic;
      adc_data_rd_clk   : out   std_logic;
      adc_data_rdy      : out   std_logic;
      adc_data_out      : out   t_adc_fifo_array;

      pll_clkout0        : out   std_logic;
      pll_clkout1        : out   std_logic;
      pll_locked         : out   std_logic;
      multi_intf_lock    : in    std_logic;
      intf_rdy           : out   std_logic;
      adc_data_overrange : out   std_logic;

      adc_io_vtc_en  : in    std_logic;
      adc_io_vtc_rdy : out   std_logic
    );
  end component adc_chan_b_intf;

  component counter_generic is
    generic (

      gbits   : positive;
      gquanta : integer
    );
    port (
      reset : in    std_logic;
[... 338 lines cut ...]

--------------------------------
-- Output squelch counter
-- Pre-load, count down to zero, stop
--------------------------------
counter_inst0 : component counter_generic
generic
map (
gbits                   => squelch_count_width,
gquanta                 => 1
)
port map (
reset                   => '0',
clk                     => adc_io_fabric_clk,
ce                      => squelch_cntr_zero_n,
clr                     => '0',
ld                      => squelch_cntr_ld,
up_dn                   => '0',
be                      => (others => '1'),
d                       => (others => '1'),
q                       => squelch_cntr_val

);

Thank you for all your help.

greenalden, 2024-07-23:
After looking into the effect of the rules blank_line_ends_group and comment_line_ends_group using your example, I realized my goal may be different than what these rules are meant for. My end goal is to have the ":" aligned across spaces and comments in the port declarations of an entity or component. Also, to have the "=>" sign aligned across comments and spaces in the port map. Perhaps this is not supported? or is there another rule I am unaware of? Thank you for all your help.

jeremiah-c-leary, 2024-07-24:
Morning @greenalden ,

I reviewed your fixed file and in the case of components is appears you would like to align colons across component declarations:
```vhdl
component one is
  port (
     port1                : in std_logic;
     port2                : in std_logic
  );
end component;

component two is
  port (
    port_name             : in std_logic;
    long_port_name        : in std_logic;
    really_long_port_name : in std_logic
  );
end component;
```
Currently the alignment rules are scoped to each component:
```vhdl
component one is
  port (
     port1 : in std_logic;
     port2 : in std_logic
  );
end component;

component two is
  port (
    port_name             : in std_logic;
    long_port_name        : in std_logic;
    really_long_port_name : in std_logic
  );
end component;
```
The same scoping applies to the component instantiations.
If I understand your desired formatting, then I would need to either create a new rule or modify the existing rule.

--Jeremy

greenalden, 2024-07-24:
Jeremy, I don't need to align across components or component port maps, but within an entity, a component, a component port map I would like the option to align across spaces and comments. I often add white space and/or a comment between signal groups for readability. Once again this was an error on my part to provide a clear enough example to show what I am trying to do. I have attached another file based on your example to hopefully better show what I am trying to do. Thank you for all your help.
[VSG_example.zip](https://github.com/user-attachments/files/16363104/VSG_example.zip)

jeremiah-c-leary, 2024-07-27:
Morning @greenalden ,

Reviewing your example, try this configuration:
```yaml
rule:
  global:
    disable: True
  comment_004:
    disable: False
  component_017:
    disable: False
    blank_line_ends_group: 'no'
    comment_line_ends_group: 'no'
  entity_017:
    disable: False
    blank_line_ends_group: 'no'
    comment_line_ends_group: 'no'
  entity_020:
    disable: False
    blank_line_ends_group: 'no'
    comment_line_ends_group: 'no'
  instantiation_010:
    disable: False
    blank_line_ends_group: 'no'
    comment_line_ends_group: 'no'
  port_007:
    disable: False
  port_008:
    disable: False
  port_009:
    disable: False
  port_020:
    disable: False
  port_map_007:
    disable: False
```

--Jeremy

greenalden, 2024-07-30:
Thank you for all the help, that seems to be what I need. Have a nice day.

closed 2024-07-30 by greenalden