jeremiah-c-leary/vhdl-style-guide #1204: Python egg for version 3.25 is broken

abyszuk, 2024-07-23:
When trying to install new version with `pip3 install vsg==3.25` I get the following error:
```
Defaulting to user installation because normal site-packages is not writeable
Collecting vsg==3.25
  Using cached vsg-3.25.0.tar.gz (1.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package vsg produced metadata for project name unknown. Fix your #egg=vsg fragments.
Discarding https://files.pythonhosted.org/packages/68/c7/4a4c39570a26f40706c999d004c587bb6ab2e13079bab8afa9520cf68caa/vsg-3.25.0.tar.gz#sha256=de3564c9e18c3cf524b585f023b0db0728b23cc79fbaf5f1fbee72f97afeebd2 (from https://pypi.org/simple/vsg/) (requires-python:>=3.8): Requested unknown from https://files.pythonhosted.org/packages/68/c7/4a4c39570a26f40706c999d004c587bb6ab2e13079bab8afa9520cf68caa/vsg-3.25.0.tar.gz#sha256=de3564c9e18c3cf524b585f023b0db0728b23cc79fbaf5f1fbee72f97afeebd2 has inconsistent name: filename has 'vsg', but metadata has 'unknown'
ERROR: Could not find a version that satisfies the requirement vsg==3.25 (from versions: 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.8.1, 0.8.2, 0.8.3, 0.8.5, 0.9, 0.10, 0.10.1, 0.11, 0.11.1, 0.11.2, 0.12, 0.12.1, 0.13, 0.14, 0.15, 0.16, 0.17, 0.18, 0.19, 0.20, 0.21, 0.22, 0.23, 0.24, 0.25, 0.26, 0.27, 0.28, 0.29, 0.30, 0.31, 0.32, 0.33, 0.34, 0.35, 0.36, 0.37, 0.38, 0.39, 0.40, 0.41, 0.42, 0.43, 1.0.0, 1.1.0, 1.2.0, 1.3.0, 1.4.0, 1.5.0, 1.6.0, 1.6.1, 1.7.0, 1.8.0, 1.8.1, 1.9.0, 1.10.0, 1.11.0, 1.11.1, 1.12.0, 2.0.0, 2.1.0, 2.2.0, 3.0.0, 3.1.0, 3.2.0, 3.2.1, 3.2.2, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.4.0, 3.5.0, 3.6.0, 3.7.0, 3.8.0, 3.9.0, 3.10.0, 3.11.0, 3.12.0, 3.12.1, 3.12.2, 3.13.0, 3.13.1, 3.14.0, 3.15.0, 3.16.0, 3.17.0, 3.18.0, 3.19.0, 3.20.0, 3.21.0, 3.22.0, 3.23.0, 3.24.0, 3.25.0)
ERROR: No matching distribution found for vsg==3.25
```
I've tested this on 3 different computers, so I don't think it's a problem on my end.

maltaisn, 2024-07-23:
Works for me.

abyszuk, 2024-07-24:
Standard programmer response :-D
No, it doesn't work either.
The only thing that works is `pip3 install --force-reinstall vsg`, but even that command generates warnings:
```
Collecting vsg
  Using cached vsg-3.25.0.tar.gz (1.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package vsg produced metadata for project name unknown. Fix your #egg=vsg fragments.
Discarding https://files.pythonhosted.org/packages/68/c7/4a4c39570a26f40706c999d004c587bb6ab2e13079bab8afa9520cf68caa/vsg-3.25.0.tar.gz#sha256=de3564c9e18c3cf524b585f023b0db0728b23cc79fbaf5f1fbee72f97afeebd2 (from https://pypi.org/simple/vsg/) (requires-python:>=3.8): Requested unknown from https://files.pythonhosted.org/packages/68/c7/4a4c39570a26f40706c999d004c587bb6ab2e13079bab8afa9520cf68caa/vsg-3.25.0.tar.gz#sha256=de3564c9e18c3cf524b585f023b0db0728b23cc79fbaf5f1fbee72f97afeebd2 has inconsistent name: filename has 'vsg', but metadata has 'unknown'
Requirement already satisfied: PyYAML>=5.1 in /usr/lib/python3/dist-packages (from vsg) (5.4.1)
```
So something in this egg is broken.

jeremiah-c-leary, 2024-07-25:
Evening,

That is odd, I get the following when I install issue `pip3 isntall vsg==3.25.0`
```mono
jcleary@DESKTOP-HV9NHA3:~$ pip3 install vsg==3.25.0
Defaulting to user installation because normal site-packages is not writeable
Collecting vsg==3.25.0
  Downloading vsg-3.25.0.tar.gz (1.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 8.4 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: PyYAML>=5.1 in ./.local/lib/python3.10/site-packages (from vsg==3.25.0) (6.0.1)
Building wheels for collected packages: vsg
  Building wheel for vsg (pyproject.toml) ... done
  Created wheel for vsg: filename=vsg-3.25.0-py3-none-any.whl size=941744 sha256=064c1e1b0a554c4cf7d72d110493fe4e157c7ea9f62c03c3060f33f83eb4f330
  Stored in directory: /home/jcleary/.cache/pip/wheels/a5/97/cd/79bf3aba107fddbffd0c07db30f8be23504e7f9cd234bb260c
Successfully built vsg
Installing collected packages: vsg
Successfully installed vsg-3.25.0

[notice] A new release of pip is available: 24.0 -> 24.1.2
[notice] To update, run: python3.10 -m pip install --upgrade pip
```
I also tried `pip3 install vsg==3.25`:
```mono
jcleary@DESKTOP-HV9NHA3:~$ pip3 install vsg==3.25
Defaulting to user installation because normal site-packages is not writeable
Collecting vsg==3.25
  Using cached vsg-3.25.0.tar.gz (1.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: PyYAML>=5.1 in ./.local/lib/python3.10/site-packages (from vsg==3.25) (6.0.1)
Building wheels for collected packages: vsg
  Building wheel for vsg (pyproject.toml) ... done
  Created wheel for vsg: filename=vsg-3.25.0-py3-none-any.whl size=941744 sha256=61eee88e7a4629faead74faa19decf09888194ad785a3cef08b56e8137271a25
  Stored in directory: /home/jcleary/.cache/pip/wheels/a5/97/cd/79bf3aba107fddbffd0c07db30f8be23504e7f9cd234bb260c
Successfully built vsg
Installing collected packages: vsg
Successfully installed vsg-3.25.0

[notice] A new release of pip is available: 24.0 -> 24.1.2
[notice] To update, run: python3.10 -m pip install --upgrade pip
```
Maybe the version of pip is the difference?

--Jeremy

jeremiah-c-leary, 2024-12-02:
Evening @abyszuk ,

I kind of lost track of this issue and wanted to check in and see if this is still a problem.

Thanks,

--Jeremy

abyszuk, 2024-12-02:
Hi,
The issue is still there for version 3.25, **but** 3.26 and 3.27 install just fine.
Must've been a glitch, I think we can close this issue.

closed 2024-12-02 by abyszuk