jeremiah-c-leary/vhdl-style-guide #1207: Documentation states incorrect option defaults for multiline indent rules

JvWesterveld, 2024-07-25:
**Environment**
```
$ vsg --version
VHDL Style Guide (VSG) version: 3.25.0
```
Installed with Pip on Ubuntu 24.04.

**Describe the bug**
[The documentation](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/3.25.0/docs/configuring_multiline_indent_rules.rst) states that the multiline indent rules have the following defaults: `yes` for option `align_left` and `no` for option `align_paren`. (These are the defaults that I prefer.) However, this documentation does not appear to be accurate as in practice the rule defaults appear to be the other way around: `no` for option `align_left` and `yes` for option `align_paren`.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `vsg -oc config`
2. Open file `config`
3. Note that the rules that have `align_left` and `align_paren` for options have values different from the documented defaults.

**Expected behavior**
The default options to match the documentation.

JvWesterveld, 2024-07-25:
Would the documentation need to be changed or the actual rule default options?

jeremiah-c-leary, 2024-07-27:
Morning @JvWesterveld ,

Good catch on the default values.  I updated the documentation to match the configuration.

I pushed an update to the `issue-1207` branch.  When you get a chance could you check it out on your end and let me know if you agree with the change?

Thanks,

--Jeremy

JvWesterveld, 2024-07-31:
Hi @jeremiah-c-leary, thanks, that branch looks good to me.

jeremiah-c-leary, 2024-09-23:
Evening @JvWesterveld,

I will get this merged to master for the next release.

--Jeremy

PR #1208 (2024-09-23): Issue#1207:  Fixing default values in multiline indent documentation.
Resolves #1207
closed 2024-09-23 by jeremiah-c-leary