jeremiah-c-leary/vhdl-style-guide #1209: Unknown reason for some words to be lowercase while other are not changed when running VSG

greenalden, 2024-07-31:
Sorry to open another question so soon after the last one but I have had some confusion around the case of some words when running VHDL through VSG. the following code has been run through VSG with no special rule set applied.

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.std_logic_arith.all;
  use ieee.std_logic_unsigned.all;

-------------------------------------------------------------------------------
-- ENTITY DECLARATION
-------------------------------------------------------------------------------

entity synchronizer is
  generic (
    channel_width : positive  := 1;
    num_stages    : integer   := 2;
    rst_value     : std_logic := '0'
  );
  port (
    rst      : in    std_logic;     -- active high reset
    clk      : in    std_logic;
    async_in : in    std_logic_vector(channel_width - 1 downto 0);
    sync_out : out   std_logic_vector(channel_width - 1 downto 0)
  );
end entity synchronizer;

-------------------------------------------------------------------------------
-- ARCHITECTURE DECLARATION
-------------------------------------------------------------------------------

architecture rtl of synchronizer is

  type stages is ARRAY (0 to num_stages - 1) OF std_logic_vector(channel_width - 1 downto 0);

  -- Registers
  signal sync_stages : stages;
  signal reset_value : std_logic_vector(channel_width - 1 downto 0) := (OTHERS => rst_value);

begin

  --------------------------------------------------------------------------
  -- Clocked Logic
  --------------------------------------------------------------------------
  process (clk, rst) is
  begin

    if (rst = '1') then

      for i IN 0 to num_stages - 1 loop

        sync_stages(i) <= reset_value;

      end loop;

      sync_out <= reset_value;
    elsif RISING_EDGE(clk) then
      sync_stages(0) <= async_in;

      for i IN 1 to num_stages - 1 loop

        sync_stages(i) <= sync_stages(i - 1);

      end loop;

      sync_out <= sync_stages(num_stages - 1);
    end if;

  end process;

end architecture rtl;
```

The part that has been confusing me is that certain things like "OF", "OTHERS", and "RISING_EDGE" have instances when they were not forced to lowercase. My understanding is that VSG would force everything to be lowercase with the default rule set. Any help would be appreciated.

jeremiah-c-leary, 2024-08-17:
Morning @greenalden ,

> The part that has been confusing me is that certain things like "OF", "OTHERS", and "RISING_EDGE" have instances when they were not forced to lowercase. My understanding is that VSG would force everything to be lowercase with the default rule set. Any help would be appreciated.

VSG defaults to keeping the users formatting unless instructed by a rule to do otherwise.  This means a rule is required to make any changes and the rules are very specific.

In your code snippet:
```vhdl
type stages is ARRAY (0 to num_stages - 1) OF std_logic_vector(channel_width - 1 downto 0);
```
The `ARRAY` and `OF` keywords would each require a rule.  The default case is lower case, as long as there is a rule for it.
 
Regards,

--Jeremy

jeremiah-c-leary, 2024-12-20:
Evening @greenalden ,

Just wanted to ping you on this issue to see if we could move it forward.

There were quite a few case rules added recently and I was wondering if they fully address the words which were not being checked for case.

Thanks,

--Jeremy

greenalden, 2024-12-20:
Jeremy, thank you for notifying me. The use case I needed VSG for has been sorted out sufficiently, at least for the time being, so I didn't need to address the problem further. It was entirely my fault for forgetting about this open issue, I should have closed it. Have a nice day.

closed 2024-12-20 by greenalden