jeremiah-c-leary/vhdl-style-guide #1211: Extend rule: assert_001 and instantiation_001

Anselmo95, 2024-08-15:
Good evening,

Currently svg only accepts this code as correct when rule assert_001 is active
```vhdl
assert WIDTH > 16
  report "FIFO width is limited to 16 bits."
  severity FAILURE;
```
I would like, if possible, to control the indentation level enforced by the rules assert_001 and instantiation_001. 
This means a way to configure the indentation for the **report** and **severity** keywords in order to consider this code as correct
```vhdl
assert WIDTH > 16
report "FIFO width is limited to 16 bits."
severity FAILURE;
```
<br/><br>
I am not exactly sure if this option should be added to the two rules I mentioned.
From a quick look around the code, trying to figure out if this could be easily done, it seems to me that this is tied to the default indentation level used by vsg. Maybe a good solution would be to expose all of this through a configuration.
In case of the assert it could be something similar to:
```yaml
indent_level:
  assert: 
      after: current
```
For the configuration example, I am taking inspiration from what i saw in the code:
- **current**: keep same indentation level
- **+1**: incremnt  indentation level
- **-1**: decremnt  indentation level

This type of configuration is a bit ugly, but it would allow for more customization in a single rule instead of adding or modifying rules for each keyword/code construct.

jeremiah-c-leary, 2024-08-17:
Morning @Anselmo95 ,

The default indentation can be overridden by including `indent_level` in your configuration.  VSG will read the default indentation configuration and then apply any changes by the user.

Just follow the same format at the default indent file.

```yaml
indent:
    tokens:
        assertion:
            keyword:
                token : current
                after : current
```

--Jeremy

Anselmo95, 2024-08-17:
Morning @jeremiah-c-leary and sorry for the bother, I somehow missed that it was already possible to change the indentation 🙄

Anyway thanks a lot for the answer and have a nice day !

closed 2024-08-17 by Anselmo95