jeremiah-c-leary/vhdl-style-guide #1213: Extend concurrent_009 to comment in between multiline condition

Anselmo95, 2024-08-23:
Good morning,

I would like if possible to keep the indentation of comments in line with the multi line condition set by rule concurrent_009.

```vhdl
output <= '1' when input = "0000" or (input = "1111" and
                                       -- comment_one
                                       input2) = "0101" else
          sig_a or sig_b when input = "0001" and
                              input = "1001" else
          sig_c and sig_d when input = "0010" else
          '0';
```
At the moment vsg is reporting  `ERROR: a.vhd(2)comment_010 -- Indent level 0` on the example above.

jeremiah-c-leary, 2024-10-13:
Evening @Anselmo95 ,

VSG makes a distinction between indenting and alignment.  Where indenting is based on hierarchy and is a set number of spaces for each level.  Alignment is the relative column locations between elements in the code.  In your example, the `output` target is at an indent, while the rest of the statement is an alignment.

I had not considered applying alignment to comments within a statement, but it makes sense since the comment is relative to the  code in the statement.

I do not believe this will be difficult, but I will need to investigate the implementation.

Thanks,

--Jeremy

Anselmo95, 2024-10-20:
Evening @jeremiah-c-leary,

Thanks for the explanation about the difference between indentation and alignment!

I have been looking at the code to see if I can implement this myself but I'm progressing quite slowly. Anyway I'll open a PR if I manage to do it.

Have a nice day!

jeremiah-c-leary, 2024-10-20:
Morning @Anselmo95 ,

I have been thinking about this and might have an idea on how to implement it.  So the indentation is set during phase 4 in `vsg/rule_list.py` lines 147 to 148:

```python
146             # Update indents before checking indent
147             if phase == 4:
148                 self.oVhdlFile.set_token_indent()
```

This function sets the indent attribute of the token using the `set_indent` method on the base token clase in `vsg/parser.py` lines 46 to 47:

```python
46     def set_indent(self, iIndent):                                                                                                                                                                            47         self.indent = iIndent                                                                                                                                                                                 
```

So maybe what is missing is an `alignment` attribute on the token class.  This attribute would be set during alignment rules.  I believe with both the `indent` and `alignment` attributes we could properly align the comments.

Using your code example:
```vhdl
|<-indent->|<--alignment------------------------->|
            output <= '1' when input = "0000" or (input = "1111" and
                                                   -- comment_one
                                                   input2) = "0101" else
                      sig_a or sig_b when input = "0001" and
                                          input = "1001" else
                      sig_c and sig_d when input = "0010" else
                      '0';
```
This would allow for indents to allow smart tabs or spaces and the alignment to be only spaces.

The base rule `vsg/rules/token_indent.py` would need to be updated to check for the `alignment` attribute.  This will probably be the majority of the effort.

The last step would be to update `vsg/rules/multiline_conditional_alignment.py`.  I want to say you can just subtract the `iFirstLine` from `iIndent` somewhere in that the rule.  This file is quite large though, but there are a lot of similarities in all the functions.  Maybe we could just add a function before the sort on line 225 that set the alignment attribute of comments?  Since another rule would actually check the indent and alignment of comments.

Regards,

--Jeremy