jeremiah-c-leary/vhdl-style-guide #1216: The ** in the description for whitespace_011 render as a URL and generates a warning when building docs.

JHertz5, 2024-08-25:
**Environment**
VSG v3.25.0

**Describe the bug**
[image]
The description for whitespace_011 includes the string "**". The restructuredtext parser recognises this as the start of a **strong text** section, and does not find an end to that strong text section. This causes it to render as a hyperlink on the docs for some reason.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://vhdl-style-guide.readthedocs.io/en/stable/whitespace_rules.html#whitespace-011
2. See error
3. Build the docs
4. See warning
```
<path>/docs/whitespace_rules.rst:201: WARNING: Inline strong start-string without end-string. [docutils]
```

**Expected behavior**
This should be rendered as plain text

JHertz5, 2024-08-25:
I've raised PR #1217 to correct this.

jeremiah-c-leary, 2024-09-09:
Evening @JHertz5 ,

Thanks for the documentation update.  I never noticed it was rendered as a link.

I will get this merged to master.

--Jeremy

PR #1217 (2024-09-09): Issue#1216: Added escape characters for "**" in docs.
Resolves #1216.
closed 2024-09-09 by jeremiah-c-leary