jeremiah-c-leary/vhdl-style-guide #1219: Corner case for combining different generate statements

larshb, 2024-08-27:
**Environment**
* VSG 3.25.0
* Ubuntu 20.04
* Python 3.8 (also tested with 3.12)

**Describe the bug**
VSG crashes when processing a specific combination of generate statements.

**To Reproduce**
Run analysis or fix on the attached snippet

**Expected behavior**
Regular execution with summary.

**Additional context**

Below is a minimal example of code/structure causing crashes (in 3.25.0)

`test_vsg.vhd`
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity test_vsg is
end entity test_vsg;

architecture sample of test_vsg is

begin

  gen0 : if (true) generate

  begin

  end generate gen0;

  gen1 : for i in 0 to 3 generate

  end generate gen1;

  gen2 : for i IN 0 to 3 generate

  end generate gen2;

  gen3 : if (true) generate

  begin

  end generate gen3;

end architecture sample;
```

`stdout`/`stderr`
```log
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/apply_rules.py", line 119, in apply_rules
    oRules.check_rules(
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/rule_list.py", line 226, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/rules/align_tokens_in_region_between_tokens.py", line 85, in analyze
    lToi = self._get_tokens_of_interest(oFile)
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/rules/align_tokens_in_region_between_tokens_when_between_tokens_unless_between_tokens.py", line 38, in _get_tokens_of_interest
    return oFile.get_tokens_bounded_by_token_when_between_tokens(self.left_token, self.right_token, self.lBetween[0], self.lBetween[1])
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/vhdlFile/vhdlFile.py", line 249, in get_tokens_bounded_by_token_when_between_tokens
    return extract.get_tokens_bounded_by_token_when_between_tokens(
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/vhdlFile/extract/get_tokens_bounded_by_token_when_between_tokens.py", line 8, in get_tokens_bounded_by_token_when_between_tokens
    lMyPairs = get_token_pairs([oLeft, oRight], [oStart, oEnd], oTokenMap)
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/vhdlFile/extract/get_tokens_bounded_by_token_when_between_tokens.py", line 28, in get_token_pairs
    lLeft, lRight = oTokenMap.get_token_pair_indexes(lInnerPair[0], lInnerPair[1])
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/token_map.py", line 61, in get_token_pair_indexes
    return extract_start_end_indexes(lStartIndexes, lEndIndexes)
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/token_map.py", line 228, in extract_start_end_indexes
    return extract_indexes_from_pairs(lPairs)
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/token_map.py", line 274, in extract_indexes_from_pairs
    if pair[0] == iStartIndex:
IndexError: list index out of range
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/larshb/.local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/home/larshb/.local/lib/python3.8/site-packages/vsg/__main__.py", line 154, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib/python3.8/multiprocessing/pool.py", line 868, in next
    raise value
IndexError: list index out of range
```

One way to "fix" the issue is by removing the `begin` in the first generate statement.

JHertz5, 2024-08-27:
It's possible that this is related to #1212.

larshb, 2024-08-28:
Seems to be solvable by requiring all `generate` to have a `begin` similar to `process`.

larshb, 2024-09-03:
@JHertz5 Looks good!

If there isn't already a rule to enforce `begin` (and `is`) for `generate`  would also be nice. Similar to `process_012` for `process`. Maybe there already is one I'm not aware of?

Closing this issue as it seems to be duplicate to #1212

closed 2024-09-03 by larshb

JHertz5, 2024-09-03:
@larshb 

> If there isn't already a rule to enforce `begin` (and `is`) for `generate` would also be nice. Similar to `process_012` for `process`. Maybe there already is one I'm not aware of?
> 
> Closing this issue as it seems to be duplicate to #1212

I also can't see any current rule to enforce `begin` or `is` for generates. If you'd like to raise feature request issues for those, I'm sure I could find some time this week to raise PRs that add those rules.

PR #1222 (2024-09-22): #Issue1212: VSG crashes when encountering generate statement corner case.
Resolves #1212
Resolves #1219
Resolves #1227

#  The problem
This bug arose in situations where there was a `for generate` with a `begin` (which is optional) followed by an `if generate` (which could be nested inside the `for generate`, it doesn't matter) with a `begin` followed by a second `for generate` without a `begin`:
```vhdl
for generate
begin
end generate;

if generate
begin
end generate;

for generate
end generate;
```

(The problem also arises if the `for generate`s are swapped for `if generate`s and vice versa [... cut]