jeremiah-c-leary/vhdl-style-guide #1223: Deprecation warnings reported during testing.

JHertz5, 2024-08-30:
**Environment**
VSG v3.25.0

**Describe the bug**
The following warnings appear during testing
```
============================================================= warnings summary ==============================================================
tests/entity/test_rule_008.py:165
  /home/jhertzog/code/vhdl-style-guide-fork/tests/entity/test_rule_008.py:165: DeprecationWarning: invalid escape sequence '\d'
    oRule.regex = "[A-Z][A-Za-z\d]*"

tests/entity/test_rule_008.py:174
  /home/jhertzog/code/vhdl-style-guide-fork/tests/entity/test_rule_008.py:174: DeprecationWarning: invalid escape sequence '\d'
    oRule.regex = "[A-Z][A-Za-z\d]*"

tests/entity/test_rule_008.py:184
  /home/jhertzog/code/vhdl-style-guide-fork/tests/entity/test_rule_008.py:184: DeprecationWarning: invalid escape sequence '\d'
    oRule.regex = "[A-Z][A-Za-z\d]*"

tests/entity/test_rule_008.py:194
  /home/jhertzog/code/vhdl-style-guide-fork/tests/entity/test_rule_008.py:194: DeprecationWarning: invalid escape sequence '\d'
    oRule.regex = "[A-Z][A-Za-z\d]*"

tests/entity/test_rule_008.py:205
  /home/jhertzog/code/vhdl-style-guide-fork/tests/entity/test_rule_008.py:205: DeprecationWarning: invalid escape sequence '\d'
    oRule.regex = "[A-Z][A-Za-z\d]*"
```

To see an example,
1. open a CI workflow
2. open 3.11-build (I haven't checked the other versions)
3. expand "build"
4. expand `tox test-py311`
5. see the warnings.

The solution to this warning can be found in this stack overflow thread https://stackoverflow.com/questions/50504500/deprecationwarning-invalid-escape-sequence-what-to-use-instead-of-d. Basically the regex patterns in tests/entity/test_rule_008.py need to be declared as raw strings instead of a Unicode strings.

JHertz5, 2024-08-30:
I've raised PR #1224 which resolves this issue.

jeremiah-c-leary, 2024-09-09:
Evening @JHertz5 ,

Thank you for tracking that down.  Time go away from me and it was pretty low on my priority list.

I will get this merged to master.

Regards,

--Jeremy

PR #1224 (2024-09-09): Issue#1223: Resolved deprecation warnings in build.
Resolves #1223.
closed 2024-09-09 by jeremiah-c-leary