jeremiah-c-leary/vhdl-style-guide #1225: vsg_rule_doc_gen script does not run

JHertz5, 2024-09-07:
**Environment**
v3.25.0

**Describe the bug**
The vsg_rule_doc_gen script runs the function `create_rule_documentation`, which has a single required argument `path`. Unfortunately, this script does not accept any arguments (other than `-h`) so it is not possible to provide a path, and the code within the `main` function for this script does not pass any argument to `create_rule_documentation`.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./bin/vsg_rule_doc_gen `
2. See error
```
$ ./bin/vsg_rule_doc_gen 
Traceback (most recent call last):
  File "/home/jhertzog/code/vhdl-style-guide-fork/./bin/vsg_rule_doc_gen", line 16, in <module>
    main()
  File "/home/jhertzog/code/vhdl-style-guide-fork/bin/../vsg/__rule_doc_gen__.py", line 24, in main
    create_rule_documentation()
TypeError: create_rule_documentation() missing 1 required positional argument: 'path'
```

**Expected behavior**
I expect the script to run and to update the documentation files.

JHertz5, 2024-09-07:
I've created PR #1226 to address this.

jeremiah-c-leary, 2024-09-09:
Evening @JHertz5 ,

That script was a intended to be part of my development flow so I did not have to update the documentation by hand.  Apparently I took a different path since it does not run.  However, `__rule_doc_gen__.py` is used to check the documentation is correct in the unit tests.

Thanks for adding the argument for the path.

--Jeremy

PR #1226 (2024-09-09): Issue-1225 vsg_rule_doc_gen script does not run.
Resolves #1225.

I've added a very rudimentary test for the vsg_rule_doc_gen script. It basically just checks that:
- The script runs without crashing.
- The script produces output files to the user-specified directory.
- The script doesn't print anything to stdout.

I haven't done any further in-depth testing since the functionality is very thoroughly tested in test_rule_doc. If you'd like the testing to be more thorough, I'd be happy to work on it a bit more, but, considering that the top-level script wasn't tested at all before, hopefully it' [... cut]
closed 2024-09-09 by jeremiah-c-leary