jeremiah-c-leary/vhdl-style-guide #1228: Create rules for subtype declarations to mirror those for type declarations

JHertz5, 2024-09-25:
**Is your feature request related to a problem? Please describe.**
There are several rules in place for types that do not exist for subtypes, even though they are equally applicable.

The rules in question that I would like to replicate for subtype declarations are as follows:
- type_006
- type_007
- type_010
- type_011
- type_017
- type_018
- type_200

Currently, if I run the default config on the following code
```vhdl
architecture rtl of test is

  type t_enum     is        record
    test : boolean;
  end record;

  subtype st_my_subtype        is      t_my_type(1 downto 0);

  type
  t_enum
  is
  record
    test : boolean;
  end record;

  subtype
  st_my_subtype
  is
  t_my_type(1 downto 0);

begin

end architecture rtl;
```
 the tool is able to fix the type declarations, but not the subtype declarations
```vhdl
architecture rtl of test is

  type t_enum is record
    test : boolean;
  end record t_enum;

  subtype st_my_subtype        is      t_my_type(1 downto 0);

  type t_enum is record
    test : boolean;
  end record t_enum;

  subtype
 st_my_subtype
  is
  t_my_type(1 downto 0);

begin

end architecture rtl;
```

**Describe the solution you'd like**
I would like rules that are available for types to be available for subtypes as well.

JHertz5, 2024-09-25:
I plan to raise a pull request for this change. I should have time to work on it this weekend.

JHertz5, 2024-09-27:
Progress

- type_006 ✅ 
- type_007 ✅ 
- type_010 ❌ 
- type_011 ❌
- type_017 ❌
- type_018 ❌
- type_200 ❌

JHertz5, 2024-09-28:
I have raised PR#1229, which is now ready for review.

jeremiah-c-leary, 2024-10-03:
Evening @JHertz5 ,

The pull request looks good.  I appreciate the tests and the documentation updates.

Question on rule `subtype_102`, did you try the `whitespace_after_token` base rule?  It should save you from having to define multiple pairs of tokens.

Regards,

--Jeremy

JHertz5, 2024-10-03:
Hi @jeremiah-c-leary. That's a great suggestion, I did not think of that. I'll try to update the PR as soon as I get a chance.
Thanks!

JHertz5, 2024-10-03:
Hi @jeremiah-c-leary,
I've implemented your suggestion and it works perfectly! That makes the code quite a bit simpler, thanks! I've also applied the same improvement to `subtype_101` by using `whitespace_after_token` (although the impact is lesser).
Thanks,
Jukka

jeremiah-c-leary, 2024-11-09:
Afternoon @JHertz5 ,

I will get this merged to master.

Thanks,

--Jeremy

closed 2024-11-09 by jeremiah-c-leary

JHertz5, 2024-11-09:
Thanks very much!

PR #1229 (2024-11-09): Issue#1228: Added subtype rules to match type rules
Resolves #1228.

| New subtype rule | Existing type rule |
| - | - |
| subtype_101 | type_006  |
| subtype_102 | type_007  |
| subtype_201 | type_010  |
| subtype_202 | type_011  |
| subtype_005 | type_017  |
| subtype_006 | type_018  |
| subtype_200 | type_200  |

I couldn't spot any more subtype rules that should be added, but I'm open to suggestions/feedback.