jeremiah-c-leary/vhdl-style-guide #1230: Documentation nitpicks

JHertz5, 2024-09-28:
I'm raising this issue to facilitate some documentation nitpick correction that I would like to make. Sorry for being pedantic, I hope this comes across as helpful rather than annoying 😅
Namely:
 - There are a few places where _it's_ is incorrectly used instead of _its_. _it's_ is a contraction of _it is_, while _its_ is the possessive form of _it_.
 - In the "classifying" documentation, mentions of the `detect` function do not put it in code formatting.
 - In the "rule selection" documentation, there are a couple of grammar fixes needed.

JHertz5, 2024-09-28:
I've raised PR #1231.

jeremiah-c-leary, 2024-10-03:
Evening @JHertz5,

> I'm raising this issue to facilitate some documentation nitpick corrections that I would like to make. Sorry for being pedantic, I hope this comes across as helpful rather than annoying

No need to apologize, I appreciate the documentation updates.  There is a lot of documentation and I am not surprised there are inconsistencies.

I will get this merged to master.

--Jeremy

PR #1231 (2024-10-03): Issue#1230: Updated documentation with nitpick corrections.
Resolves #1230.
closed 2024-10-03 by jeremiah-c-leary