jeremiah-c-leary/vhdl-style-guide #1232: Capitalisation rules for type identifiers do not work on types that are declared in a different file.

JHertz5, 2024-09-28:
**Environment**
VSG v3.26.0

**Describe the bug**
The rule _type_004_ enforces capitalisation of type identifiers. This rule appears to be triggered by the type declaration. As a result, this means that types declared elsewhere, e.g. in a package that is a separate file, or that is part of the standard library, such as `character` are not corrected by this rule.

I suspect that the latter example is the reason that the rule `ieee_500` was introduced. This rule solves the problem for a few predefined types, namely:
- `std_logic_vector`
- `std_ulogic_vector`
- `std_ulogic`
- `std_logic`
- `integer`
- `signed`
- `unsigned`

but it leaves out many more, inclding `natural` which is defined by the parser, but isn't included in the list of tokens that _ieee_500_ acts upon. My point is that if we could try to add every predefined type, but that just adds more code and puts us at a greater risk of introducing human errors.

I suggest that, unless there is another reason that these keywords are treated specially, we treat them as any other type declaration and handle the capitalisation of type and subtype names in a more generic way.

One thing that we could do is replace rule _ieee_005_ with a single rule for _type_mark_ tokens. That way, _type_004_ and _subtype_002_ will continue to handle types and subtype with local declarations, and the new rule, _type_mark_500_, (or perhaps a better name), could handle type marks for type declared elsewhere, including VHDL's predefined types.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file "test.vhd" with the following contents:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
  use ieee.math_real.all;

entity test is
end entity test;

architecture rtl of test is

  type my_type is range 10 to 100;

  subtype my_subtype   is MY_TYPE range 10 to 20;
  subtype my_subtype_t is MY_EXTERNAL_TYPE range 10 to 20;
  subtype my_int       is INTEGER range 10 to 100;
  subtype my_nat       is NATURAL range 10 to 100;
  subtype my_pos       is POSITIVE range 10 to 100;
  subtype my_char      is CHARACTER range 'a' downto 'b';

  signal s_my_subtype   : MY_EXTERNAL_SUBTYPE;
  signal s_my_subtype   : MY_SUBTYPE;
  signal s_my_subtype_t : MY_SUBTYPE_T;
  signal s_my_int       : MY_INT;
  signal s_my_nat       : MY_NAT;
  signal s_my_pos       : MY_POS;
  signal s_my_char      : MY_CHAR;
  signal s_my_type_t    : MY_TYPE_T;
  signal s_int          : INTEGER;
  signal s_nat          : NATURAL;
  signal s_pos          : POSITIVE;
  signal s_char         : CHARACTER;

begin

end architecture rtl;
```
2. Run `vsg -f test.vhd --fix`
3. Observe test.vhd:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
  use ieee.math_real.all;

entity test is
end entity test;

architecture rtl of test is

  type my_type is range 10 to 100;

  subtype my_subtype   is my_type range 10 to 20;
  subtype my_subtype_t is MY_EXTERNAL_TYPE range 10 to 20;
  subtype my_int       is integer range 10 to 100;
  subtype my_nat       is NATURAL range 10 to 100;
  subtype my_pos       is POSITIVE range 10 to 100;
  subtype my_char      is CHARACTER range 'a' downto 'b';

  signal s_my_subtype   : MY_EXTERNAL_SUBTYPE;
  signal s_my_subtype   : my_subtype;
  signal s_my_subtype_t : my_subtype_t;
  signal s_my_int       : my_int;
  signal s_my_nat       : my_nat;
  signal s_my_pos       : my_pos;
  signal s_my_char      : my_char;
  signal s_my_type_t    : MY_TYPE_T;
  signal s_int          : integer;
  signal s_nat          : NATURAL;
  signal s_pos          : POSITIVE;
  signal s_char         : CHARACTER;

begin

end architecture rtl;
```
4. Note that `MY_EXTERNAL_TYPE `, `NATURAL`, `POSITIVE`, and `CHARACTER` are still upper case. The latter three are all predefined VHDL types.

**Expected behavior**
I expect the tool to enforce capitalisation on all type marks consistently.

JHertz5, 2024-09-28:
I will raise a PR with my suggested changes so that we can see what it looks like. I understand that this change in approach may not be agreed with (or may prove to have some errors that I have not forseen) so we can always reject the change if it doesn't work out.

JHertz5, 2024-09-28:
I've created a PR #1233 to resolve this. I'm open to any feedback/rejection. One potential issue to consider is overlap with _type_014_ and _subtype_002_ - maybe those rules can be deprecated if we prefer this approach?

jeremiah-c-leary, 2024-12-09:
Evening @JHertz5 ,

I'm not convinced this quite the implementation we would want in this case.  There is probably a way to merge `ieee_500`, `type_014` and `subtype_002` into a single `type_mark` rule.  Almost like a `consistent_token_case` rule with exceptions.  So the type and subtype case can be maintained and for those types not known, we could apply a proper case.   It could be as simple as filling out the exception on line 133 in file `consistent_case_utils.py`:
```python
109 def create_tois(lAllDicts, oFile):                                                                110     lReturn = []
111     oTokenMap = oFile.get_token_map()
112     for dDict in lAllDicts:
113         for iName in dDict["names"]:
114             try:
...
133             except KeyError:
134                 pass
135
136     return lReturn
```
and updating consistent_token_case to allow for the usual `token_case` options.  Maybe `consistent_token_case` inherits `token_case` instead of `case.Rule`.

What do you think?

--Jeremy

JHertz5, 2024-12-11:
Hi @jeremiah-c-leary

> There is probably a way to merge ieee_500, type_014 and subtype_002 into a single type_mark rule. Almost like a consistent_token_case rule with exceptions. So the type and subtype case can be maintained and for those types not known, we could apply a proper case.

That's a very interesting suggestion, I hadn't considered this. I like the sound of that a lot, I'll give it a try!

Thanks very much,
Jukka

jeremiah-c-leary, 2024-12-11:
Afternoon @JHertz5 ,

> If I'm understanding correctly, this is a single rule that would enforce consistent case on type marks that have a local declaration, and enforce proper case on type marks that do not have a local declaration to be consistent with.

That is correct.

> On the other hand, I suppose that the alternative solution is to leave type_014 and subtype_002 (i.e. the consistent case functionality) as they are and make this type mark rule act only on type marks that don't have a definition (i.e. doing only the proper case functionality).

That is a good point. We would be making a complex rule when two simple ones would suffice.  The key would be ensuring the `_get_tokens_of_interest` only returns those type marks without local definitions.

What if we pushed lines 41 through 96 of `consistent_token_case.py` into `cc_utils.py`.  Then we add a function called `get_non_local_type_marks` to `cc_utils.py`.  We could use the two functions in the `_get_tokens_of_interest` in the new rule so we only get those tokens that are not defined locally.

--Jeremy

JHertz5, 2024-12-15:
Hi @jeremiah-c-leary 

I've implemented your suggestion, I now have `type_mark_500` applied only to type_marks that are not locally declared. I think that it is working well. I have one more issue though. Currently, the type marks in port clauses or generic clauses are not handled by the rule, to avoid overlap with `generic_017` and `port_018`. However, those two rules only apply to a select few IEEE type names, meaning that there is no rule to enforce case on other type marks in those clauses. Do you have any preferences on how we approach this? My gut feeling is that `type_mark_500` should cover all type marks, and `generic_017` and `port_018` would be deprecated, but I'd be keen to know if you have an opinion on the matter.

Thanks,
Jukka

jeremiah-c-leary, 2024-12-18:
Afternoon @JHertz5 ,

> I have one more issue though. Currently, the type marks in port clauses or generic clauses are not handled by the rule, to avoid overlap with generic_017 and port_018. However, those two rules only [...]

I had not thought about those `type_mark`s.  I believe the best option is to have a rule for a production and only add exceptions if a user requests it.  Essentially we assume the user would want the same formatting of a production regardless of it's location in the file.

For example, if we take the `signal` keyword in the `signal_declaration` production:
```mono
signal_declaration ::= signal identifier_list : sub_type_indication [ signal_kind ] [ := expression ] ;
```
We should assume the user would want `signal` lower or uppercase in every production where a `signal_declaration` can be used:
```mono
block_declaration_item
entity_declaration_item
interface_object_declaration
object_declaration
package_declarative_item
```
> My gut feeling is that type_mark_500 should cover all type marks, and generic_017 and port_018 would be deprecated. This feels right to me since then, there is just one rule that covers everything. However, I'd be keen to know if you have an opinion on the matter. 

I would 100% agree.  Do you think it would be possible to include changing `generic_017` and `port_018` in this issue?

Thanks,

--Jeremy

JHertz5, 2024-12-18:
Hi @jeremiah-c-leary

Excellent, thanks very much. Absolutely, I'll make those changes. I will update the PR when I get a chance, probably at the weekend at the latest.

Thanks,
Jukka

JHertz5, 2024-12-20:
Hi @jeremiah-c-leary

I've updated the branch so that it now deprecates `generic_017` and `port_018`, and covers those type_marks with the new rule `type_mark_500`.

I have realised that the consistent case subtype and type rules do not work in packages, and as a result, the new rule, which borrows from the consistent case rules, also does not work with packages. I feel like this PR already involves a lot of changes so I'd be more inclined to break this problem out to a separate issue, but I can try to fix it in the current PR if you'd prefer.

Thanks,
Jukka

jeremiah-c-leary, 2025-01-01:
Morning @JHertz5 ,

> I have realised that the consistent case subtype and type rules do not work in packages, and as a result, the new rule, which borrows from the consistent case rules, also does not work with [...]

I think we have taken this PR as far as we should.  I would prefer to address the package problem in a different issue.

I will get this merged to master.

Thanks again,

--Jeremy

closed 2025-01-01 by jeremiah-c-leary

JHertz5, 2025-01-01:
Excellent, thanks very much! I've raised #1384 to capture the consistent case changes.

PR #1233 (2025-01-01): Issue#1232: Replaced ieee_500 with type_mark_500
Resolves #1232.

Potential issues:
- ~~Conflict/overlap with `type_014` and `subtype_002`. Those rules enforce _consistent_ capitalisation, rather than strictly upper or lower case, so we would lose the ability to allow for more complex case conventions if we disable them. However, from the new rule, we gain the ability to enforce casing on more type marks. For the purposes of this PR, I have assumed that we will get around this problem by setting `type_014` and `subtype_002` to be disabled by default, but I am very open to feedback on this.~~ T [... cut]