jeremiah-c-leary/vhdl-style-guide #1234: Consistent capitalization of generic names doesn't work inside generic maps

thgbe, 2024-09-30:
**Environment**
VSG 3.26.0 on Ubuntu 22.04

**Describe the bug**
Rule architecture_600 (consistent capitalization of **generic** names in an architecture body) doesn't apply inside generic maps, while architecture_601 (consistent capitalization of **port** names in an architecture body) does inside port maps.

**To Reproduce**
The following file has all uppercase ports, generics, and signals:
```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  generic (
    G1 : integer := 0;
    G2 : integer := 0
  );
  port (
    P1 : in    std_logic;
    P2 : out   std_logic
  );
end entity test;

architecture test of test is

  component test2 is
    generic (
      G1 : integer := 0;
      G2 : integer := 0
    );
    port (
      P1 : in    std_logic;
      P2 : out   std_logic
    );
  end component test2;

  signal S1 : std_logic;
  signal S2 : std_logic;

begin

  test2_i : component test2
    generic map (
      G1 => G1,
      G2 => G2
    )
    port map (
      P1 => P1,
      P2 => S1
    );

  test3_i : entity work.test3
    generic map (
      G1 => G1,
      G2 => G2
    )
    port map (
      P1 => P1,
      P2 => S2
    );

  P2 <= P1 when G1 = 0 and G2 = 0 else
        '0';

end architecture test;
```
With default settings, it is fixed to:
```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  generic (
    g1 : integer := 0;
    g2 : integer := 0
  );
  port (
    p1 : in    std_logic;
    p2 : out   std_logic
  );
end entity test;

architecture test of test is

  component test2 is
    generic (
      g1 : integer := 0;
      g2 : integer := 0
    );
    port (
      p1 : in    std_logic;
      p2 : out   std_logic
    );
  end component test2;

  signal s1 : std_logic;
  signal s2 : std_logic;

begin

  test2_i : component test2
    generic map (
      g1 => G1,
      g2 => G2
    )
    port map (
      p1 => p1,
      p2 => s1
    );

  test3_i : entity work.test3
    generic map (
      g1 => G1,
      g2 => G2
    )
    port map (
      p1 => p1,
      p2 => s2
    );

  p2 <= p1 when g1 = 0 and g2 = 0 else
        '0';

end architecture test;
```
All port, signal, and generic definitions are changed to lowercase. Their appearances in the architecture are, too, with the exception of generics use on the right side of generic maps. This is true for direct and component instantiation.

JHertz5, 2024-09-30:
Hi @thgbe.
Thanks for raising this. It looks like rule `architecture_601` uses a completely different (and obviously improved) set of code compared to `architecture_600`, despite their functionality being very similar. I've been able to basically copy and paste the new code into `architecture_600` and change it to target generics instead of ports to get your example code correctly formatted, but it'll need a bit more work to tidy up the code, refactor out common code between the two rules, etc, and get it ready to be merged. In the meantime, you are welcome to try out the code on my branch at https://github.com/JHertz5/vhdl-style-guide/tree/issue-1234.

JHertz5, 2024-10-03:
I've raised PR #1235 to resolve this issue. I'm open to any feedback on how it could be improved.

jeremiah-c-leary, 2024-12-09:
Evening @JHertz5 ,

PR looks good.  I will merge this to master.

--Jeremy

closed 2024-12-09 by jeremiah-c-leary

JHertz5, 2024-12-10:
Thanks very much!

PR #1235 (2024-12-09): Issue#1234: Consistent capitalization of generic names doesn't work inside generic maps
Resolves #1234.
I've basically copied and pasted a bunch of code across from `architecture_601` to a common base rule that includes `architecture_600`. Very open to feedback, especially on naming, code structure, etc.