jeremiah-c-leary/vhdl-style-guide #1236: Rule for case in hexadecimal constants and base specifier

thgbe, 2024-10-04:
**Is your feature request related to a problem? Please describe.**
Our coding guideline prescribes upper case for letters in hexadecimal numeric constants and lower case for the "x" before the hex string. I cannot find a rule that allows enforcing this:

Before:
`constant c : std_logic_vector(31 downto 0) := X"cafebabe";`

Required after:
`constant c : std_logic_vector(31 downto 0) := x"CAFEBABE";`

**Describe the solution you'd like**
Add a rule that allows enforcing the case of letters in hexadecimal numeric constants (`extended_digit`) and, separately, the `base_specifier`.

JHertz5, 2024-10-05:
Hi @thgbe. I've had a go at fulfilling your request. I still need to work on it a bit more to pass all of the CI tests, but in the meantime, could you please check whether this branch https://github.com/JHertz5/vhdl-style-guide/tree/issue-1236 works for you?

thgbe, 2024-10-07:
Thank you very much for the quick implementation @JHertz5! I tried you branch, and it works nicely for me.

PR #1237 (2024-10-08): Issue#1236: Rule for case in hexadecimal constants and base specifier
Resolves #1236.

Potential issues to consider for the reviewer:
- The new rules are part of the case group, but they are not part of any case subgroup. From what I can tell, all other case rules are part of a subgroup, but it didn't seem like these rules fit into any of those groups.
- I've set the default value of rule_501 to "upper". I'm not sure whether there are any other case rule where the default is "upper",  if this is not preferred, we can change the default to match the other case rules.
- It's my first attempt at  [... cut]
closed 2024-10-08 by jeremiah-c-leary