jeremiah-c-leary/vhdl-style-guide #1241: Reduce duplication in classification unittests

JHertz5, 2024-10-06:
**Is your feature request related to a problem? Please describe.**
There is a great deal of duplicated code in the test files in tests/vhdlFile/, which test the parser's ability to classify various tokens. For example, here is a diff between a few pairs of files:

```diff
$ diff -u test_aggregate.py test_access_type_definition.py
--- test_aggregate.py   2024-08-25 09:58:55.027345900 +0100
+++ test_access_type_definition.py      2024-08-25 09:58:55.027345900 +0100
@@ -2,8 +2,8 @@
 import os
 import unittest
 
+import vsg.vhdlFile as vhdlFile
 from tests import utils
-from vsg import vhdlFile
 
 sLrmUnit = utils.extract_lrm_unit_name(__name__)
```
The only difference there is a slight variation in the import of vhdlFile.

```diff
$ diff -u test_aggregate.py test_case_generate_statement.py 
--- test_aggregate.py   2024-08-25 09:58:55.027345900 +0100
+++ test_case_generate_statement.py     2024-10-06 18:15:33.191458133 +0100
@@ -5,7 +5,7 @@
 from tests import utils
 from vsg import vhdlFile
 
-sLrmUnit = utils.extract_lrm_unit_name(__name__)
+sLrmUnit = "case_generate_statement"
 
 lFile, eError = vhdlFile.utils.read_vhdlfile(os.path.join(os.path.dirname(__file__), sLrmUnit, "classification_test_input.vhd"))
 oFile = vhdlFile.vhdlFile(lFile)
```
The only difference there is that one file uses a hard-coded string rather than using the function that extracts the string from the file name.

All of the files are either exactly the same, or have very minor variations like those above. This is a lot of duplication. I am raising this issue as I would like to try to reduce this duplication.

**Describe the solution you'd like**
I've found a [post on stackoverflow](https://stackoverflow.com/a/20870875) which I believe would fit this situation very nicely. It requires no new python modules and no changes to the testing methodology that is currently in place.

I will try to raise a PR that solves this issue.

JHertz5, 2024-10-06:
Here is a diff of all of the .py files in the directory as evidence of the code duplication. The files that have significant differences will not be removed in this change.
<details>
 <summary>Log</summary>

```diff
--- __init__.py	2024-08-25 09:58:55.007345902 +0100
+++ test_todo.py	2024-08-25 09:58:55.027345900 +0100
@@ -0,0 +1,27 @@
+# -*- coding: utf-8 -*-
+import os
+import unittest
+
+from tests import utils
+from vsg import vhdlFile
+
+sLrmUnit = utils.extract_lrm_unit_name(__name__)
+
+lFile, eError = vhdlFile.utils.read_vhdlfile(os.path.join(os.path.dirname(__file__), sLrmUnit, "classification_test_input.vhd"))
+oFile = vhdlFile.vhdlFile(lFile)
+
+
+class test_token(unittest.TestCase):
+    def test_classification(self):
+        self.maxDiff = None
+        sTestDir = os.path.join(os.path.dirname(__file__), sLrmUnit)
+
+        lExpected = []
+        utils.read_file(os.path.join(sTestDir, "classification_results.txt"), lExpected, False)
+
+        lActual = []
+
+        for oObject in utils.extract_objects(oFile, True):
+            lActual.append(str(oObject))
+
+        self.assertEqual(lExpected, lActual)
--- test_access_type_definition.py	2024-10-06 18:36:28.691575201 +0100
+++ test_todo.py	2024-08-25 09:58:55.027345900 +0100
@@ -2,8 +2,8 @@
 import os
 import unittest
 
-import vsg.vhdlFile as vhdlFile
 from tests import utils
+from vsg import vhdlFile
 
 sLrmUnit = utils.extract_lrm_unit_name(__name__)
 
@@ -13,6 +13,7 @@
 
 class test_token(unittest.TestCase):
     def test_classification(self):
+        self.maxDiff = None
         sTestDir = os.path.join(os.path.dirname(__file__), sLrmUnit)
 
         lExpected = []
--- test_aggregate.py	2024-10-06 18:36:28.691575201 +0100
+++ test_todo.py	2024-08-25 09:58:55.027345900 +0100
@@ -13,6 +13,7 @@
 
 class test_token(unittest.TestCase):
     def test_classification(self):
+        self.maxDiff = None
         sTestDir = os.path.join(os.path.dirname(__file__), sLrmUnit)
 
         lExpected = []
--- test_alias_declaration.py	2024-10-06 18:36:28.691575201 +0100
+++ test_todo.py	2024-08-25 09:58:55.027345900 +0100
@@ -2,8 +2,8 @@
 import os
 import unittest
[... 1017 lines cut ...]
-sLrmUnit = "use_clause"
+sLrmUnit = utils.extract_lrm_unit_name(__name__)
 
 lFile, eError = vhdlFile.utils.read_vhdlfile(os.path.join(os.path.dirname(__file__), sLrmUnit, "classification_test_input.vhd"))
 oFile = vhdlFile.vhdlFile(lFile)
@@ -13,6 +13,7 @@
 
 class test_token(unittest.TestCase):
     def test_classification(self):
+        self.maxDiff = None
         sTestDir = os.path.join(os.path.dirname(__file__), sLrmUnit)
 
         lExpected = []
--- test_variable_declaration.py	2024-08-25 09:58:55.027345900 +0100
+++ test_todo.py	2024-08-25 09:58:55.027345900 +0100
@@ -13,6 +13,7 @@
 
 class test_token(unittest.TestCase):
     def test_classification(self):
+        self.maxDiff = None
         sTestDir = os.path.join(os.path.dirname(__file__), sLrmUnit)
 
         lExpected = []
```
</details>

JHertz5, 2024-10-06:
I have raised PR #1242 to resolve this issue.

jeremiah-c-leary, 2024-10-13:
Evening @JHertz5 ,

That is a very elegant solution to the problem.  I never knew you could do such a thing in `unittest`.

Thank you for the test improvement.

I will get this merged to master.

--Jeremy

PR #1242 (2024-10-13): Issue#1241: Reduced duplication in token classification tests
Resolves #1241.

This PR replaces a set of explicitly defined tests that share identical code with a single test that generates test cases for each test. The list of subdirectories in the directory is used to build the list of test cases.

It is possible to use the list of classifier source files as the source of the test cases instead. The advantage of this would be that, when a new token is introduced, it will automatically have a test case generated for it, and the test case will fail until the classification input and result are [... cut]
closed 2024-10-13 by jeremiah-c-leary