jeremiah-c-leary/vhdl-style-guide #1243: Parsing fails for external names pathnames that include a parenthesis not at the end.

JHertz5, 2024-10-08:
**Environment**
VSG's current master (at time of writing)
git hash:
```
ce86655850d0e24df420e3799b9b5036596a820d
```

**Describe the bug**
An external pathnames can have parenthesis in them, for example if there is a `for generate` statement in the path. At the moment, the external names rules are run on a path that includes a parenthesis with anything after the parenthesis, VSG fails with an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file "test.vhd" with the following contents:
```vhdl
architecture rtl of test is

begin

  count_output <= << signal .gen(gv_counter).o_count : t_count >>;

end architecture rtl;
```
2. Run `vsg -f test.vhd`
3. Observe the output
```
Error while processing test.vhd: 
Error: Unexpected token detected while parsing external_signal_name @ Line 5, Column 41 in file None
       Expecting : :
       Found     : .o_count
```

**Expected behavior**
I expect the tool to be able to function normally on this kind of statement.

JHertz5, 2024-10-08:
I have found the source of the problem and I plan to raise a pull request to address it soon.

JHertz5, 2024-10-08:
I've created PR #1244 to resolve this issue.

jeremiah-c-leary, 2024-10-13:
Morning @JHertz5 ,

Thanks for the pull request.

I have not used external names before so it is challenging to create testcases.  It would be nice if there was some standard set of files that could be used to run parsers against.

I will get this merged to master.

--Jeremy

PR #1244 (2024-10-13): Issue#1243: Fixed external names parsing bug
Resolves #1243.

I've solved the problem by iteratively checking for opening parentheses at the end of the current portion of the pathname,e .g .
```vhdl
probe_signal <= << constant some.hierarchy.hier(i).test(i).my_loop(i).test : std_logic >>;
-- while loop checks at these points           ^       ^          ^       ^
```
It's possible that an alternative solution would be to absorb the parentheses into the pathname token? But perhaps the parentheses are separated to allow for multiline indentation rules and we need to keep them separate. Happy to  [... cut]
closed 2024-10-13 by jeremiah-c-leary