jeremiah-c-leary/vhdl-style-guide #1245: Regex impossible to match (with TerosHDL)

ArnePret, 2024-10-09:
**Environment**
VSG 3.26.0
Windows 11
TerosHDL v6.0.10 (with VSG selected as VHDL formatter)

**Describe the bug**
When trying to Configuring Uppercase and Lowercase Rules the regex option appears to be doing nothing.
Currently a lot of prefix_exceptions are defined in my Style.yaml, which all work fine.
It would be a lot less work though, to write one line of regex and be done with it.
But regex never matches anything for me.

**To Reproduce**
Install VSCode
Install TerosHDL (follow instructions on their website)
Configure VSG as selected VHDL formatter
Link yaml stylesheet

In Style.yaml:
  architecture_013:
    case: 'regex'
    regex:
      = '[A-Za-z]*'

In vhdl file:
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
  use ieee.math_real.all;

entity Test is
end entity Test;

architecture arch of Test is

begin

end architecture arch;

**Expected behavior**
Accept arch as correct, since it matches the regex.

**Screenshots**
[image]

jeremiah-c-leary, 2024-10-13:
Morning @ArnePret and @JHertz5 ,

@JHertz5 is correct that the `=` should be removed.  However, it looks like the documentation is incorrect:
[image]

It shows the equal sign in the example.   I will update the documentation.

Regards,

--Jeremy

JHertz5, 2024-10-13:
Hi @jeremiah-c-leary. Thanks for pointing that out, I had no idea!

The pull request that you've raised corrects the regex field, but there are other mistakes in the snippet. Perhaps we should correct all of them in this PR?
The code in the PR is
```
prefix_exceptions: - 'G_' suffix_exceptions: - '_G' case_exceptions: - 'IEEE' regex: ''
```
I don't have access to my computer at the moment, but if I'm not mistaken, the `-` after each of the `:`s is erroneous and should be removed.

jeremiah-c-leary, 2024-10-13:
Evening @JHertz5 ,

The dash after the colon in YAML indicates a list.  The `prefix_expections`, `suffix_exceptions`, and `case_exceptions` allow for multiple entries, while `regex` does not.  You can double check this by looking at the definitions in the `token_case` class:

```python
  9 class token_case(case.Rule):
 10     """
 11     Checks the case for words.
 12
 13     Parameters
 14     ----------
 15
 16     name : string
 17        The group the rule belongs to.
 18
 19     identifier : string
 20        unique identifier.  Usually in the form of 00N.
 21
 22     lTokens : list of token types
 23        The token types to check the case on.
 24     """
 25
 26     def __init__(self, lTokens):
 27         super().__init__()
 28         self.solution = None
 29         self.phase = 6
 30         self.case = "lower"
 31         self.configuration.append("case")
 32         self.lTokens = lTokens
 33         self.prefix_exceptions = []
 34         self.suffix_exceptions = []
 35         self.case_exceptions = []
 36         self.regex = ""
 37         self.oRegex = None
```

Regards,

--Jeremy

JHertz5, 2024-10-14:
Hi @jeremiah-c-leary. Ah my mistake, thanks for the correction!

jeremiah-c-leary, 2024-10-15:
I will merge this to master.

--Jeremy

closed 2024-10-15 by jeremiah-c-leary

ArnePret, 2024-10-17:
Hi @JHertz5 and @jeremiah-c-leary,

Thank you both for the clarification and update to the documentation.

PR #1246 (2024-10-15): Issue#1245:  Fixing documentation issue around regex.
Resolves #1245