jeremiah-c-leary/vhdl-style-guide #1252: Case rules for directions in formal parameter declarations for functions and procedures

JHertz5, 2024-10-15:
**Is your feature request related to a problem? Please describe.**
At the moment, there are no rules to enforce the case of direction keywords (`in`, `out`, `inout`) in formal parameter declarations in functions and procedures.

**Describe the solution you'd like**
I would like a rule to enforce the case of these keywords, e.g. turning
```vhdl
architecture rtl of test is

  procedure my_proc (
    PARAM1 : IN integer;
    PARAM2 : OUT integer
  ) is
  begin

  end procedure my_proc;

  function my_func (
    PARAM1 : IN integer;
    PARAM2 : OUT integer
  ) return integer is
  begin

  end function my_func;

begin

end architecture rtl;
```
to
```vhdl
architecture rtl of test is

  procedure my_proc (
    PARAM1 : in integer;
    PARAM2 : out integer
  ) is
  begin

  end procedure my_proc;

  function my_func (
    PARAM1 : in integer;
    PARAM2 : out integer
  ) return integer is
  begin

  end function my_func;

begin

end architecture rtl;
```

JHertz5, 2024-10-15:
I'm planning to raise a PR to address this.

JHertz5, 2024-10-15:
Use [port_019](https://vhdl-style-guide.readthedocs.io/en/latest/port_rules.html#port-019) as an example.

JHertz5, 2024-10-18:
PR #1260 raised.

jeremiah-c-leary, 2024-10-20:
Evening @JHertz5 ,

Another perfect PR.

It never ceases to amaze me how much there is format.

I will get this merged to master.

Thanks again,

--Jeremy

closed 2024-10-20 by jeremiah-c-leary

JHertz5, 2024-10-20:
Thanks very much! Yes, it's quite incredible haha

PR #1260 (2024-10-20): Issue#1252: Rules for enforcing case on function and procedure parameter directions
Resolves #1252.
The numbering of the rules is skipping a few numbers to avoid conflicts with other branches that I'm working on.