jeremiah-c-leary/vhdl-style-guide #1253: Case rule for the pure/impure keyword in function declarations

JHertz5, 2024-10-15:
**Is your feature request related to a problem? Please describe.**
There are currently no rules to enforce the case of the `pure`/`impure` keyword in function declarations.

**Describe the solution you'd like**
I'd like a rule that would enforce the case on these keywords, e.g. changing
```vhdl
architecture rtl of test is

  IMPURE function my_func (
    PARAM1 : IN integer;
    PARAM2 : OUT integer
  ) return integer is
  begin

  end function my_func;

  PURE function my_func (
    PARAM1 : IN integer;
    PARAM2 : OUT integer
  ) return integer is
  begin

  end function my_func;

begin

end architecture rtl;
```
into
```vhdl
architecture rtl of test is

  impure function my_func (
    PARAM1 : IN integer;
    PARAM2 : OUT integer
  ) return integer is
  begin

  end function my_func;

  pure function my_func (
    PARAM1 : IN integer;
    PARAM2 : OUT integer
  ) return integer is
  begin

  end function my_func;

begin

end architecture rtl;
```

JHertz5, 2024-10-15:
I'm planning to raise a PR to address this.

JHertz5, 2024-10-18:
I've raised PR #1259 to address this issue.

jeremiah-c-leary, 2024-10-20:
Morning @JHertz5 ,

Could we split this into two rules.  One for `pure` and another for `impure`.  The user may want to set either to upper or lower case.

Thanks,

--Jeremy

JHertz5, 2024-10-20:
Hi @jeremiah-c-leary

That's a very good point, thanks for the feedback. I will update the PR soon.

Thanks,
Jukka

jeremiah-c-leary, 2024-11-09:
Afternoon @JHertz5 ,

The PR looks good.  I am going to merge this to master.

Thanks again,

--Jeremy

closed 2024-11-09 by jeremiah-c-leary

JHertz5, 2024-11-09:
Thanks very much!

PR #1259 (2024-11-09): Issue#1253: Added rule to enforce case of `pure` and `impure` keywords.
Resolves #1253.
The rule numbering skips a few numbers to avoid conflictsfor other rules that I'm working on.