jeremiah-c-leary/vhdl-style-guide #1255: Prefix Exception not Respected in Rule use_clause_501

JennySmith888, 2024-10-17:
**Environment**
Installed using `pip install vsg --use-pep517 ` on Ubuntu 22.04.4 LTS

**Describe the bug**
I made the rule:
```
 use_clause_501:
   case: 'PascalCase'
   prefix_exceptions: ['std_']
```
But when I run `vsg`, it reports errors with packages that begin with `std_`

**To Reproduce**
Steps to reproduce the behavior:
1. see `AxiVersionTb_badsyntax.vhd` and `my_config.yaml` pasted at the end (GitHub does not support attaching these filetypes)
2. run `vsg --configuration my_config.yaml --quality_report report.yaml -f AxiVersionTb_badsyntax.vhd`

Observe the errors:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  use_clause_501            | Error      |         16 | Format std_logic_1164 into PascalCase
  use_clause_501            | Error      |         17 | Format std_logic_unsigned into PascalCase
  use_clause_501            | Error      |         18 | Format std_logic_arith into PascalCase
```
View the generated `quality_report.yaml` if desired.

**Expected behavior**
I expect the rule `use_clause_501` to be enforced for all package names except those that begin with `std_`. The packages that begin with `std_` should not be reformatted and should not throw an error but should remain as they are in all [lowercase.]([url]())

Full `my_config.yaml`

```
---
rule:
  global:
    indent_size: 3

  attribute_declaration_501:
      case: 'PascalCase'

  attribute_declaration_502:
      case: 'PascalCase'

  block_500:
      case: 'PascalCase'

  entity_008:
      case: 'PascalCase'

  entity_012:
      case: 'PascalCase'

  entity_014:
     case: 'lower'

  architecture_004:
      case: 'lower'

  architecture_011:
      case: 'lower'

  architecture_013:
      case: 'lower'

  architecture_014:
      case: 'PascalCase'

  instantiation_008:
      disable: true

  instantiation_009:
     case: 'PascalCase'

  instantiation_028:
    case: 'PascalCase'
    prefix_exceptions : ['ieee']

  instantiation_034:
    disable: true

  package_008:
      case: 'PascalCase'

  package_010:
     case: 'PascalCase'

  package_body_502:
    case: 'PascalCase'

  package_body_507:
    case: 'PascalCase'

  port_010:
    case: 'camelCase'

  port_map_002:
    case: 'camelCase'

  library_500:
    case: 'lower'

  use_clause_500:
    case: 'lower'

  use_clause_501:
    case: 'PascalCase'
    prefix_exceptions: ['std_'] 

  constant_004:
    case: 'upper'

  constant_600:
    disable: false
    suffixes: ['_C']

  component_008:
    case: 'PascalCase'

  component_012:
    case: 'PascalCase'

  generic_600:
    disble: false
    suffixes: ['_G']

  generic_007:
    case: 'upper'

  generic_017:
    case: 'lower'

  generic_map_002:
    case: 'upper'

  signal_004:
    case: 'camelCase'

  signal_007:
    disable: true

  process_017:
    case: 'lower'

  process_019:
    case: 'lower'

  variable_004:
    case: 'camelCase'

  variable_007:
    disable: true

indent:
  tokens:
    use_clause:
      keyword:
        token_after_library_clause: '+0'
        token_if_no_matching_library_clause: '+0'
...

```

Full ` AxiVersionTb_badsyntax.vhd`:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_unsigned.all;
use ieee.std_logic_arith.all;

library surf;
use surf.StdRtlPkg.all;
use surf.AxiLitePkg.all;

library ruckus;
use ruckus.BuildInfoPkg.all;

entity AxiVersionTb is
end entity AxiVersionTb;

architecture testbed of AxiVersionTb is

   constant GET_BUILD_INFO_C : BuildInfoRetType   := toBuildInfo(BUILD_INFO_C);
   constant MOD_BUILD_INFO_C : BuildInfoRetType   :=
   (
      buildString => GET_BUILD_INFO_C.buildString,
      fwVersion   => GET_BUILD_INFO_C.fwVersion,
      gitHash     => x"1111_2222_3333_4444_5555_6666_7777_8888_9999_AAAA"
   );  -- Force githash
   constant SIM_BUILD_INFO_C : slv(2239 downto 0) := toSlv(MOD_BUILD_INFO_C);

   constant CLK_PERIOD_G : time := 10 ns;
   constant TPD_G        : time := CLK_PERIOD_G / 4;

   signal axilClk         : sl                     := '0';
   signal axilRst         : sl                     := '0';
   signal axilWriteMaster : AxiLiteWriteMasterType := AXI_LITE_WRITE_MASTER_INIT_C;
   signal axilWriteSlave  : AxiLiteWriteSlaveType  := AXI_LITE_WRITE_SLAVE_INIT_C;
   signal axilReadMaster  : AxiLiteReadMasterType  := AXI_LITE_READ_MASTER_INIT_C;
   signal axilReadSlave   : AxiLiteReadSlaveType   := AXI_LITE_READ_SLAVE_INIT_C;

begin

   --------------------
   -- Clocks and Resets
   --------------------
   U_axilClk : entity surf.ClkRst
      generic map (
         CLK_PERIOD_G      => CLK_PERIOD_G,
         RST_START_DELAY_G => 0 ns,
         RST_HOLD_TIME_G   => 1000 ns
      )
      port map (
         clkP => axilClk,
         rst  => axilRst
      );
```

jeremiah-c-leary, 2024-10-18:
Evening @JennySmith888,

Could you try the following change to your configuration:
```yaml
rule:
    use_clause_501:
        case: 'PascalCase'
        case_exceptions:
            - 'std_logic_1164'
            - 'std_logic_unsigned'
            - 'std_logic_arith'
```

That will enforce lower case on the three standard logic libraries while enforcing PascalCase on all other libraries.

Regards,

--Jeremy

JennySmith888, 2024-10-18:
Great, that worked, thanks! It would be helpful if there was a way to make the exception more general so it doesn not need to be fully-defined for each standard logic library. Perhaps using something like a wild character, ex. `std_*` could work and be intuitive. I can submit a feature request for this is desired. Thanks for the quick help!

jeremiah-c-leary, 2024-10-19:
Morning @JennySmith888 ,

> Great, that worked, thanks! 

Awesome.

> It would be helpful if there was a way to make the exception more general so it doesn not need to be fully-defined for each standard logic library. Perhaps using something like a wild character, ex. std_* could work and be intuitive.

The exception was implemented to enforce a case for a specific string.  So the exception does not exclude the rule from running.

If we allowed wildcards, e.g. `std_*`, then any of the following would pass the rule:
`std_LoGiC_1164`
`std_LOGiC_1164`

Which I assume would not be desired.

I may be missing a valid use case here though.  Could you elaborate on the behavior you would expect if a wild card was allowed?

Regards,

--Jeremy

jeremiah-c-leary, 2024-11-09:
Afternoon @JennySmith888 ,

Just wanted to ping you on this issue to see if we could move it forward.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-12-02:
Evening @JennySmith888 ,

Just wanted to ping you on this issue to see if we could move it forward.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-12-20:
Evening @JennySmith888 ,

Just wanted to ping you on this issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2025-01-04:
Happy New Year @JennySmith888 ,

Just wanted to ping you on this issue.

Thanks,

--Jeremy

jeremiah-c-leary, 2025-01-18:
Morning @JennySmith888 ,

I am going to assume this is has been resolved and will be closing the issue.  If there is additional work to be done, please reopen this issue or create a new one.

Regards,

--Jeremy

closed 2025-01-18 by jeremiah-c-leary