jeremiah-c-leary/vhdl-style-guide #1256: clkP is incorrectly considered not camelCase by rule port_map_002

JennySmith888, 2024-10-17:
**Environment**
Installed using `pip install vsg --use-pep517 ` on Ubuntu 22.04.4 LTS

**Describe the bug**
I made the rule:
```
 port_map_002:
   case: 'camelCase'
```
But when I run `vsg`, it reports an error with a port named `clkP`

**To Reproduce**
Steps to reproduce the behavior:
1. see `AxiVersionTb_badsyntax.vhd` and `my_config.yaml` pasted at the end (GitHub does not support attaching these filetypes)
2. run `vsg --configuration my_config.yaml --quality_report report.yaml -f AxiVersionTb_badsyntax.vhd`

Observe the error:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  port_map_002              | Error      |         63 | Format clkP into camelCase
----------------------------+------------+------------+--------------------------------------
```
View the generated `quality_report.yaml` if desired.

**Expected behavior**
I expect the rule `port_map_002` to not throw an error for `clkP` not being camelCase because it is camelCase?

Full `my_config.yaml`:

```
---
rule:
  global:
    indent_size: 3

  attribute_declaration_501:
      case: 'PascalCase'

  attribute_declaration_502:
      case: 'PascalCase'

  block_500:
      case: 'PascalCase'

  entity_008:
      case: 'PascalCase'

  entity_012:
      case: 'PascalCase'

  entity_014:
     case: 'lower'

  architecture_004:
      case: 'lower'

  architecture_011:
      case: 'lower'

  architecture_013:
      case: 'lower'

  architecture_014:
      case: 'PascalCase'

  instantiation_008:
      disable: true

  instantiation_009:
     case: 'PascalCase'

  instantiation_028:
    case: 'PascalCase'
    prefix_exceptions : ['ieee']

  instantiation_034:
    disable: true

  package_008:
      case: 'PascalCase'

  package_010:
     case: 'PascalCase'

  package_body_502:
    case: 'PascalCase'

  package_body_507:
    case: 'PascalCase'

  port_010:
    case: 'camelCase'

  port_map_002:
    case: 'camelCase'

  library_500:
    case: 'lower'

  use_clause_500:
    case: 'lower'

  use_clause_501:
    case: 'PascalCase'
    prefix_exceptions: ['std_'] 

  constant_004:
    case: 'upper'

  constant_600:
    disable: false
    suffixes: ['_C']

  component_008:
    case: 'PascalCase'

  component_012:
    case: 'PascalCase'

  generic_600:
    disble: false
    suffixes: ['_G']

  generic_007:
    case: 'upper'

  generic_017:
    case: 'lower'

  generic_map_002:
    case: 'upper'

  signal_004:
    case: 'camelCase'

  signal_007:
    disable: true

  process_017:
    case: 'lower'

  process_019:
    case: 'lower'

  variable_004:
    case: 'camelCase'

  variable_007:
    disable: true

indent:
  tokens:
    use_clause:
      keyword:
        token_after_library_clause: '+0'
        token_if_no_matching_library_clause: '+0'
...

```

Full ` AxiVersionTb_badsyntax.vhd`:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_unsigned.all;
use ieee.std_logic_arith.all;

library surf;
use surf.StdRtlPkg.all;
use surf.AxiLitePkg.all;

library ruckus;
use ruckus.BuildInfoPkg.all;

entity AxiVersionTb is
end entity AxiVersionTb;

architecture testbed of AxiVersionTb is

   constant GET_BUILD_INFO_C : BuildInfoRetType   := toBuildInfo(BUILD_INFO_C);
   constant MOD_BUILD_INFO_C : BuildInfoRetType   :=
   (
      buildString => GET_BUILD_INFO_C.buildString,
      fwVersion   => GET_BUILD_INFO_C.fwVersion,
      gitHash     => x"1111_2222_3333_4444_5555_6666_7777_8888_9999_AAAA"
   );  -- Force githash
   constant SIM_BUILD_INFO_C : slv(2239 downto 0) := toSlv(MOD_BUILD_INFO_C);

   constant CLK_PERIOD_G : time := 10 ns;
   constant TPD_G        : time := CLK_PERIOD_G / 4;

   signal axilClk         : sl                     := '0';
   signal axilRst         : sl                     := '0';
   signal axilWriteMaster : AxiLiteWriteMasterType := AXI_LITE_WRITE_MASTER_INIT_C;
   signal axilWriteSlave  : AxiLiteWriteSlaveType  := AXI_LITE_WRITE_SLAVE_INIT_C;
   signal axilReadMaster  : AxiLiteReadMasterType  := AXI_LITE_READ_MASTER_INIT_C;
   signal axilReadSlave   : AxiLiteReadSlaveType   := AXI_LITE_READ_SLAVE_INIT_C;

begin

   --------------------
   -- Clocks and Resets
   --------------------
   U_axilClk : entity surf.ClkRst
      generic map (
         CLK_PERIOD_G      => CLK_PERIOD_G,
         RST_START_DELAY_G => 0 ns,
         RST_HOLD_TIME_G   => 1000 ns
      )
      port map (
         clkP => axilClk,
         rst  => axilRst
      );
```

JennySmith888, 2024-10-17:
According to the regex defined [here ](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/db548fdbdb7fd0649e32335f2febbaf236e56b79/vsg/rules/case_utils.py#L7), `clkP` should be considered camelCase.
```💀  ~ ipython3
Python 3.10.12 (main, Sep 11 2024, 15:47:36) [GCC 11.4.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.31.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import re

In [2]: camelCase = re.compile("(?:[a-z])+(?:[a-z0-9])*((?:[A-Z])+(?:[a-z0-9])+)*")

In [3]:  bool(re.match(camelCase, 'clkP'))
Out[3]: True
```

JHertz5, 2024-10-18:
Hi @JennySmith888. Thanks for raising this issue.

# Problem

> According to the regex defined [here ](https://github.com/jeremiah-c-leary/vhdl-style-guide/blob/db548fdbdb7fd0649e32335f2febbaf236e56b79/vsg/rules/case_utils.py#L7), clkP should be considered camelCase.

I don't agree with this statement. In fact, I think that this is the source of the problem that you are experiencing. The regex is quite long and complex, so to explain my reasoning better, here is the regexwith the groups separated and annotated:
```
(?:[a-z])+      # lowercase letters (1 or more)
(?:[a-z0-9])*   # lowercase letters and numbers (0 or more)
(
  (?:[A-Z])+    # uppercase letters (1 or more)
  (?:[a-z0-9])+ # lowercase letters and numbers (1 or more). The problem is here
)*              # (0 or more)
```
This regex will not capture `clkP`, because it does not have a lowercase letter or number after the capital letter. The names `clkPa` and `clkP0` do match because they have a lowercase letter or number after the capital letter.

## Testing
### Source file
```vhd

architecture rtl of test is

begin

  cmp_test : entity my_cmp
    port map (
      clkPa => x,
      clkP0 => y,
      clkP  => z
    );

end architecture rtl;
```
### Config file:
```yaml
rule:
  port_map_002:
    case: 'camelCase'
  instantiation_034:
    disable: True
```
### Output
```
$ bin/vsg -f test.vhd -c test.yml
================================================================================
File:  test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 770
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  port_map_002              | Error      |         10 | Format clkP into camelCase
----------------------------+------------+------------+--------------------------------------
```

# Solution
 If we change the regex so that there is a `*` rather than a `+` after that problematic group, then `clkP` does match because the letter/number after the capital letter is now optional.

## Testing
I've modified line 7 in case_utils.py as follows:
```diff
- camelCase = re.compile("(?:[a-z])+(?:[a-z0-9])*((?:[A-Z])+(?:[a-z0-9])+)*")
+ camelCase = re.compile("(?:[a-z])+(?:[a-z0-9])*((?:[A-Z])+(?:[a-z0-9])*)*")
```
### Output
```
$ bin/vsg -f test.vhd -c test.yml
================================================================================
File:  test.vhd
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 773
Total Violations:    0
  Error   :     0
  Warning :     0
```

I agree that `clkP` should count as camel case. The current regex also would fail if the string ends with a string of captial letters, so an acronym at the end of the name, e.g. `enableFIFO`, would fail, which also seems wrong. I'll raise a PR to merge the change described above, and do a similar change for the PascalCase regex.

JHertz5, 2024-10-18:
@JennySmith888 Another thing - if you need a "quick fix" and don't want to wait for the PR to be raised, merged and released, you can use the `case: 'regex'` option instead and use the regex mentioned above.

JHertz5, 2024-10-18:
I've tried to find an "official" definition for camelCase and PascalCase, and I've stumbled across this [stack overflow answer](https://stackoverflow.com/a/47591707) that references the Google java style guide. This seems as official as I'm likely to find, so I'll use these:
```
camelCase:
[a-z]+((\d)|([A-Z0-9][a-z0-9]+))*([A-Z])?
breakdown:
[a-z]+             # lowercase letter (1 or more)
(                  
    (\d)           # number
    |              # or
    (
        [A-Z0-9]   # number or uppercase letter (1)
        [a-z0-9]+) # number or lowercase letter (1 or more)
)*                 # (1 or more)
([A-Z])?           # uppercase letter (0 or 1)

PascalCase:
([A-Z][a-z0-9]+)((\d)|([A-Z0-9][a-z0-9]+))*([A-Z])?
breakdown:
(
    [A-Z]         # uppercase letter (1)
    [a-z0-9]+     # number or lowercase letter (1 or more)
)(
    (\d)          # number
    |             # or
    (
        [A-Z0-9]  # number or uppercase letter (1)
        [a-z0-9]+ # number or lowercase letter (1 or more)
    )
)*                # (1 or more)
([A-Z])?          # uppercase letter (0 or 1)
```
Note that this camelCase does allow @JennySmith888's `clkP` name, so this change will resolve this issue.

JHertz5, 2024-10-18:
Hi @JennySmith888, I've create PR #1257 to resolve this issue. If you could check that you're happy with the changes, that would be very helpful. The regex for camelCase is now:
```diff
- (?:[a-z])+(?:[a-z0-9])*((?:[A-Z])+(?:[a-z0-9])+)*
+ (?:[a-z])+(?:[a-z0-9])*((?:[A-Z])+(?:[a-z0-9])+)*([A-Z])?
```

JennySmith888, 2024-10-18:
Yes that new regex in PR https://github.com/jeremiah-c-leary/vhdl-style-guide/pull/1257 is working great for me. Thanks for the quick fix!

JHertz5, 2024-10-18:
Excellent! No problem, glad that I could help!

jeremiah-c-leary, 2024-10-20:
Afternoon @JennySmith888  and @JHertz5 ,

> I've tried to find an "official" definition for camelCase and PascalCase, and I've stumbled across this [stack overflow answer](https://stackoverflow.com/a/47591707) that references the Google java style guide. This seems as official as I'm likely to find, so I'm thinking to use these:

Too bad I did not find that stack overflow answer back when camelCase was added.

Thanks for the PR @JHertz5 ,  I will get it merged to master.

--Jeremy

PR #1257 (2024-10-20): Issue#1256: Updated PascalCase and camelCase regexes to allow a single uppercase letter at the end of the name
Resolves #1256.
closed 2024-10-20 by jeremiah-c-leary