jeremiah-c-leary/vhdl-style-guide #1263: Rules to enforce case on parameter class in procedures and functions

JHertz5, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
I would like to have rules to enforce case on parameter classes (e.g. `signal`, `constant`, etc.) in functions and procedures.
For example
```vhdl
architecture rtl of test is

  procedure overflow (
    a          : integer;
    VARIABLE a : integer;
    CONSTANT a : integer;
    SIGNAL   a : integer;
    FILE     a : file_type
  ) is
  begin

  end procedure overflow;

begin

end architecture rtl;
```
I would like a rule to change the parameter classes to lowercase.

JHertz5, 2024-10-18:
I am planning to raise a PR to address this. FYI @urbite.

jeremiah-c-leary, 2024-10-20:
Morning @JHertz5 ,

Another perfect PR.

I will merge this to master.

Thanks again,

--Jeremy

closed 2024-10-20 by jeremiah-c-leary

JHertz5, 2024-10-20:
Thanks very much!

PR #1264 (2024-10-20): Issue#1263: Rules for enforcing case on parameter class in functions and parameters
Resolves #1263.
The rule numbering skips a few numbers to avoid conflicts with other active branches.