jeremiah-c-leary/vhdl-style-guide #1267: Update rule return_statement_300 to handle return labels

JHertz5, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
Return statements can have labels, but rule return_statement_300 does not correct indentation if there is a label.

For example, given the following code, the standalone return statement's indentation is corrected, but the labelled return statement's indentation is not.
```vhdl
  function func1 return integer is begin

return;
my_label : return;

  end function func1;
```
Correction:
```vhdl
  function func1 return integer is begin

    return;
my_label : return;

  end function func1;
```

JHertz5, 2024-10-18:
I've raised PR #1268 to resolve this issue.

jeremiah-c-leary, 2024-10-20:
Morning @JHertz5 ,

I'm thinking we should consider the following case with respect to indenting:
```
label : 
    return;
```
I believe that would require the label indent to be a separate rule and then the `after` setting in the indent yaml file to be set to `+1`.

What do you think?

--Jeremy

JHertz5, 2024-10-20:
Hi @jeremiah-c-leary

You raise a good point, I agree that this would be desirable. Now that I'm looking at it again, it also doesn't feel right for the label to piggyback on the return keyword's rule. I'll update the PR. There are a few similar cases (null statement, exit statement) that I'll raise a separate issue for.

Thanks very much
Jukka

jeremiah-c-leary, 2024-10-20:
I just realized the label for the return statement is optional, which is making me second guess my suggestion.  If the label is there then the `after` indent of the semicolon would have to be '-2', but if the label is not there then the `after` indent of the semicolon will need to be `-1`.    The example I gave is not handled by other optional labels, so I would suggest we keep the indents as `current` for the label.

> ...it also doesn't feel right for the label to piggyback on the return keyword's rule

I would agree.

Thanks,

Jeremy

JHertz5, 2024-10-21:
Hi @jeremeiah-c-leary

Ok, that makes sense. In that case, I will split the rule to handle the optional label separately, but will leave the indentation config the same.

JHertz5, 2024-10-21:
Hi @jeremiah-c-leary

I've updated the PR to treat labels separately in a new rule.

Thanks,
Jukka

jeremiah-c-leary, 2024-11-09:
Afternoon @JHertz5 ,

PR looks good.  I will merge it to master.

--Jeremy

closed 2024-11-09 by jeremiah-c-leary

JHertz5, 2024-11-09:
Thanks very much!

PR #1268 (2024-11-09): Issue#1267: Updated rule return_statement_300 to handle label alignment
Resolves #1267.