jeremiah-c-leary/vhdl-style-guide #1269: Rules to enforce case on exit statements

JHertz5, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
I'd like rules to enforce case on exit statements, e.g. correcting
```vhdl
      label_exit : EXIT my_loop when condition;
      EXIT my_loop when condition;
      EXIT when condition;
      EXIT;
```
to
```vhdl
      label_exit : exit my_loop when condition;
      exit my_loop when condition;
      exit when condition;
      exit;
```

JHertz5, 2024-10-18:
I've raised PR #1270 to resolve this issue. FYI @urbite.

jeremiah-c-leary, 2024-10-20:
Morning @JHertz5 ,

This PR looks also look perfect.

I will get this merged to master.

Thanks again,

--Jeremy

closed 2024-10-20 by jeremiah-c-leary

JHertz5, 2024-10-20:
Thanks very much!

> BTW, after this many pull requests do you have any feedback on the whole process? Anything from updating the code base to the GitHub process that could be improved?

@jeremiah-c-leary On the whole, I've found the process of raising these PRs to be very smooth and easy. I'm very grateful for your codebase being so modular and well organised. Especially for the case rules that I've been adding recently, it is very simple to copy another rule, tweak a few bits and be done (which is why I've been spamming so many PRs this week!)

The only part of the process that feels unnecessarily manual for me is part of the documentation updates. I use the vsg_rule_doc_gen tool to update the rule documentation, which is really helpful and saves me lots of time, but this doesn't update the documentation for rule groups (e.g. the case::keyword rule group), configuration (e.g. configuring_uppercase_and_lowercase_rules), or the list of different rule sets at the top-level in rules.rst. Since the test for rule links doesn't tell you which file failed the test (as far as I can tell), there is sometimes a bit of friction trying to work out where to make corrections if I've made a mistake.

I would like to see if I can update vsg_rule_doc_gen to automate more of the documentation writing process (or perhaps there is already an automated way to do this that I'm not aware of!)

PR #1270 (2024-10-20): Issue#1269: Rule to enforce case on exit statements
Resolves #1269.