jeremiah-c-leary/vhdl-style-guide #1277: Rule to enforce case on `others` keyword in case statements

JHertz5, 2024-10-20:
**Is your feature request related to a problem? Please describe.**
I'd like a rule to enforce the case on the `others` keyword in case statements, e.g.
```vhdl
    case expression is

      when choice =>

        a <= '1';

      when OTHERS =>

        b <= x"FFFF";

    end case;
```
corrected to
```vhdl
    case expression is

      when choice =>

        a <= '1';

      when OTHERS =>

        b <= x"FFFF";

    end case;
```
This should be similar to `case_generate_alternative_501`.

JHertz5, 2024-10-20:
It seems that, while case-generate rules are split between `case_generate_statement` and `case_generate_alternative`, the rules for regular case statements are all within `case`. I'll add the new rule to `case` for consistency.

jeremiah-c-leary, 2024-10-20:
Evening @JHertz5,

> It seems that, while case-generate rules are split between case_generate_statement and case_generate_alternative, the rules for regular case statements are all within case. I'll add the new rule to case for consistency.

You can tell which rules were generated before I sat down and really thought about the proper method to name the rules.  I like the current method of using the production name, just wish I would have thought about it earlier.

I will get this merged to master.

I really appreciate your efforts to add rules and improve this tool.  I have had several interactions with people at CERN and the Square Kilometer Array (SKA) telescope project.  I would assume VSG is still being used in those locations.  It would be awesome to work at CERN, but this is the closest I will get, but still...to be able to contribute in a small way to science is satisfying.  Now that you are contributing, you are also helping out those projects...assuming you do not work for either already.

Anecdotally, I know that SpaceX also use[s|d] VSG.  I had a friend go for an interview and they were talking about linting tools.  The guys at SpaceX said they use VSG, and my friend said "I know that guy that wrote that."

Again, just wanted to say thank you for the effort you are putting in.

Regards,

--Jeremy

closed 2024-10-20 by jeremiah-c-leary

urbite, 2024-10-21:
@jeremiah-c-leary, @JHertz5 
Just want to say thanks and much props for your efforts on vsg:  @jeremiah-c-leary  for creating and maintaining it and @JHertz5 for being so responsive in addressing issues I've found during first use of vsg.

I'm looking forward to learning how to more fully utilize vsg. Hopefully I'll be able to contribute back to the project in the future.

JHertz5, 2024-10-21:
Hi @jeremiah-c-leary 

Thanks very much for merging and for your kind words. That is extremely cool to hear about, the thought of making a (very very small) contribution to those projects is amazing! I just work for a satcom startup in the UK. As someone who strongly believes in the value of consistent and clear code formatting, VSG has been very useful for me and has saved me a lot of effort while writing and reviewing code. I've benefitted greatly from countless free and open source tools like this one and I'm very glad to be able to "give back" somewhat (plus writing some Python is a fun change of pace from the day job). 

Thanks again!
Jukka

JHertz5, 2024-10-21:
@urbite You're very welcome! Notifying us about bugs/missing features is a valuable contribution in itself, so thanks to you as well. If you run into any more problems or have any questions, don't hesitate to open another issue and we'll do our best to help.

urbite, 2024-10-21:
> @urbite You're very welcome! Notifying us about bugs/missing features is a valuable contribution in itself, so thanks to you as well. If you run into any more problems or have any questions, don't hesitate to open another issue and we'll do our best to help.

From prior discussion you've noted that there are quite a number of keywords that don't currently have case rules. Since you're aware of them, should I still report words that I come across that aren't yet supported? If so, is this best done by discussion or opening a feature request issue?

The latest 4 keywords I'm using with no case rules: **impure, protected, shared, postponed**. Just used these for the first time, hence the new discovery.

PR #1278 (2024-10-20): Issue#1277: Rule for enforcing case on `others` in case statements.
Resolves #1277.
 There is already rule to cover [others keywords in case generate alternatives](https://vhdl-style-guide.readthedocs.io/en/latest/case_generate_alternative_rules.html#case-generate-alternative-501) so this PR only addresses case statement alternatives.