jeremiah-c-leary/vhdl-style-guide #1280: Rules to enforce case on the `others` keyword in selected assignments

JHertz5, 2024-10-20:
**Is your feature request related to a problem? Please describe.**
I'd like a rule to enforce case on the `others` keyword in selected assignments. For example, I'd like
```vhdl
architecture rtl of fifo is

begin

  my_proc : process (all) is
  begin

    with CONDITION select A <=
      '1' when X,
      '0' when OTHERS;

    with CONDITION select A <=
      '1' when X,
      '0' when OTHERS;

  end process my_proc;

end architecture rtl;
```
corrected to
```vhdl
architecture rtl of fifo is

begin

  my_proc : process (all) is
  begin

    with CONDITION select A <=
      '1' when X,
      '0' when others;

    with CONDITION select A <=
      '1' when X,
      '0' when others;

  end process my_proc;

end architecture rtl;

JHertz5, 2024-10-20:
This suffers from the same complication described here https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/1279#issuecomment-2424766551
There are multiple constructs covered by the `selected_assignment` rules, i.e. 
- `concurrent_selected_signal_assignment`
- `selected_force_assignment`
- `selected_variable_assignment`
- `selected_waveform_assignment`
At the moment, it looks like I'll need to extend or create a variant of `token_case_in_range_bounded_by_tokens` to accept lists of pairs of start and end tokens.

jeremiah-c-leary, 2024-12-12:
Evening @JHertz5 ,

I did a search for `choice` in the productions and found the following:

[image]

I also ran `vsg_parser` on your example and the `others` keyword is part of the `choice` production:
```mono
--------------------------------------------------------------------------------
14 |       '0' when OTHERS;
<class 'vsg.parser.character_literal'>
<class 'vsg.token.selected_waveforms.when_keyword'>
<class 'vsg.token.choice.others_keyword'>
<class 'vsg.token.selected_waveform_assignment.semicolon'>
```

We could have a single rule to set the case for the `others` in a `choice` production.

What do you think?

--Jeremy

JHertz5, 2024-12-12:
Hi @jeremiah-c-leary 

I think that's an excellent idea! It simplifies the ruleset, it fits the LRM well, and it makes the implementation trivial. I'll put a PR together for this soon.

Thanks,
Jukka

JHertz5, 2024-12-12:
I've created a PR (#1346) to resolve this issue. This creates a new rule `choice_500` as suggested by @jeremiah-c-leary, and deprecates a few rules that were enforcing case on specific instances of `choice` `others` keywords.

jeremiah-c-leary, 2024-12-13:
Morning @JHertz5 ,

The PR looks perfect.  It nice that it reduced the number of rules.

I will get this merged to master.

Thanks,

--Jeremy

PR #1346 (2024-12-13): Issue#1280: Added rule to enforce case of `others` keyword in `choice` productions
Resolves #1280.
I've also deprecated a number of rules that are now covered by the new rule.