jeremiah-c-leary/vhdl-style-guide #1286: VSG changing file permissions

ALL-SPACE-Anuj, 2024-10-22:
**Environment**
VHDL Style Guide (VSG) version: 3.26.0
Rocky Linux 8.9

**Describe the bug**
Permissions of files change after running VSG.

**To Reproduce**
Run VSG
Check changes to permissions with git.
You should see the permissions change to "-rwxr-x---"

**Expected behavior**
Expected no change to permissions.

jeremiah-c-leary, 2024-11-10:
Morning @ALL-SPACE-Anuj ,

I ran a test case on a file 

```mono
$ ls -lart deleteme.vhd
-rw-r--r-- 1 jcleary jcleary 14 Nov 10 08:05 deleteme.vhd
```

with the following command `cp deleteme.vhd deleteme2.vhd;vsg -f deleteme2.vhd --fix` and I get the following:

```mono
$ ls -lart deleteme*.vhd
-rw-r--r-- 1 jcleary jcleary 14 Nov 10 08:05 deleteme.vhd
-rw-r--r-- 1 jcleary jcleary 14 Nov 10 08:08 deleteme2.vhd
```

I'm wondering if your `umask` is forcing the change in file permissions.

--Jeremy

ALL-SPACE-Anuj, 2024-11-28:
Thanks for getting back to me.

My umask is set to 022, so default directory permissions are 755 and default file permissions are 644.

I was a bit vague/confused in my first comment so here's a little more detail:
.sh and .tcl files are not affected (it is a VHDL style tool after-all, I must have messed that up on my own somehow...)

Changing .vhd file permissions to 755, 777, or 000, and then running VSG changes file permissions to 644.

It's possible this could be because of the umask. Either way, 644 is fine with me and I don't have permissions to play with the umask so I'll stop digging here.

jeremiah-c-leary, 2024-11-29:
Morning @ALL-SPACE-Anuj ,

> My umask is set to 022, so default directory permissions are 755 and default file permissions are 644.

ah...now I understand.  I just ran an experiment and set the file permission to 755, ran VSG and checked permissions:
```mono
$ ls -lart blah*
-rwxr-xr-x 1 jcleary jcleary 15 Nov 29 09:30 blah.vhd
-rw-r--r-- 1 jcleary jcleary 15 Nov 29 09:30 blah2.vhd
```
and I see the permission change.

Let me check on any options when creating a file.  I assumed the permissions would be maintained.

Regards,

--Jeremy

jeremiah-c-leary, 2024-11-29:
Morning @ALL-SPACE-Anuj ,

I pushed an update to the `issue-1286` branch that should fix the permission changing issue.  When you get a chance could you check it out and let me know if it is working for you?

Thanks,

--Jeremy

ALL-SPACE-Anuj, 2024-12-02:
On master:
[image]
On issue-1286:
[image]

Looks good to me.

jeremiah-c-leary, 2024-12-06:
Awesome, I will merge this to master.

--Jeremy

PR #1342 (2024-12-06): Issue#1286:  Ensuring existing file permissions are maintained on fixed files.
Resolves #1286
closed 2024-12-06 by jeremiah-c-leary